MHeironimus/ArduinoJoystickLibrary #1: HTML5 Gamepad compability

Sebring, 2015-12-17:
Hi MHeironimus, 
thank you for this very easy to use and straightforward library. 

This is probably more of a question than an issue, I hope you don't mind.

I've successfully used you library, when testing my 1-axis joystick (a Leonardo with a b10k pot) in Win 7 everything works perfect. It register -127 to 127 and goes edge to edge.

But here's the issue, it does not work properly as a HTML5 Gamepad. According to the [HTML5 spec](https://developer.mozilla.org/en-US/docs/Web/API/Gamepad_API/Using_the_Gamepad_AP) it expect axis value range from -1 to 1. I was under the impression that this range conversion would be handled by the browser but it seems the browser receives the -127 to 127 values.
[Gampad API tester](http://www.html5rocks.com/en/tutorials/doodles/gamepad/gamepad-tester/tester.html)
[HTML5 Gamepad Tester](http://html5gamepad.com/)

Do you, or any one else of course, have any idea how to approach this to make the joystick comply to the HTML5 specs?

I have [the question on stackoverflow](http://stackoverflow.com/questions/34323062/arduino-joystick-to-web-gamepad).

MHeironimus, 2015-12-17:
I will look into that issue and get back with you. 

What browser are you using?

Sebring, 2015-12-17:
Thank you, 
I've yet only tested in Chrome (47) and Chrome Canary (49). I might have tested in Firefox but I'm honestly not sure. I'll test on FF this evening (GMT+1).

MHeironimus, 2015-12-17:
I loaded an Arduino Leonardo up with the UsbJoystickTest.ino sketch and used the following page to test in Chrome  47 [Version 47.0.2526.80 m (64-bit) and Version 47.0.2526.106 m (64-bit)], Firefox 42.0, and Firefox 43.0 on Windows 7: http://www.html5rocks.com/en/tutorials/doodles/gamepad/gamepad-tester/tester.html

It seemed to be working correctly. Would you be able to post a simple sample that demonstrates the issue you are seeing?

Thanks,
Matt

MHeironimus, 2015-12-17:
I was not able to reproduce the issue using the Joystick.h library, but I was able to reproduce it using the Joystick3.h library. I will dig into it some more and see if I can figure out what is going on.

Sebring, 2015-12-17:
Thank you Matthew.
I was indeed using Joystick2.h, sorry I should have mentioned. I'm kind of embarrassed of the fact that I obviously never tested in Firefox before starting all this. However, I guess it's good to know of Chromes current shortcomings of the gamepad.

Thanks again Matthew, feel free to close this issue.

_sidenote:_
I will use your library to build _'Atari 2600 Paddles'_ for my retro game project /Sebring/OlympicsRR

MHeironimus, 2015-12-17:
Thanks for bringing the issue up. I had not tested my library using any browsers, so it was interesting to investigate what was working and what was not.

I would be curious to see your Arduino Sketch file and see how you are wiring up the Atari 2600 paddles to the Arduino.

closed 2015-12-17 by MHeironimus

Sebring, 2015-12-17:
I'm glad you see it that way.

The 'paddles' will get their own repo one day but as for now I included the arduino code in the readme for [OlympicsRR](https://github.com/Sebring/OlympicsRR). At the time writing the paddles are two b10k pots mounted on a suitable cardbox with the Leonardo inside the cardbox, hehe.