MHeironimus/ArduinoJoystickLibrary #2: After running the test my computer wont find the device

coolmon48, 2016-01-01:
I have an Arduino Leonardo and I downloaded the single joystick library. After putting the file in my libraries I ran the test, and it uploaded to my board fine, but when I go to devices to find my board its not there. So I opened my device manager and the board is showing up there, but still not in the devices. I have tried to unplug the board and plug it back in and reset the board, but nothing works. In the HID properties it says "Device started" "Device configured", and then it says "Device not migrated." I am using windows 10 i'm not sure if that matters. Please let me know if you can help me at all.

Sebring, 2016-01-01:
Sorry, but I have to ask since it's not menitoned. You are using Arduino IDE 1.6.6 right, and not an older version?

coolmon48, 2016-01-01:
yes its 1.6.6, but I actually just restarted my computer and now it is working. Thanks.

closed 2016-01-01 by MHeironimus