MHeironimus/ArduinoJoystickLibrary #3: A gamepad support

GAMELASTER, 2016-02-22:
Hello,

I want to make from my Arduino Pro Micro (Atmega32u4) to a gamepad (2 axis thumb joystick) and some buttons. Any ideas or documentations of gamepad HID?

Thanks

MHeironimus, 2016-02-22:
Jakkul has a good example of how he used the library on his blog at [http://jakkul.blogspot.com/2015/06/sim-racing-and-game-controllers-part-2.html](http://jakkul.blogspot.com/2015/06/sim-racing-and-game-controllers-part-2.html). I also have an example on my blog at [http://mheironimus.blogspot.com/2015/12/ultimate-classic-game-console-joystick.html](http://mheironimus.blogspot.com/2015/12/ultimate-classic-game-console-joystick.html).

GAMELASTER, 2016-02-22:
@MHeironimus
Thanks! But anyway, i still have problem with HID Descriptor.
I used a reference: http://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/
```
static const uint8_t ReportDescriptor[48] = {
    0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
    0x09, 0x05,                    // USAGE (Game Pad)
    0xa1, 0x01,                    // COLLECTION (Application)
    0x85, 0x03,                    //   REPORT_ID (3)
    0xa1, 0x00,                    //   COLLECTION (Physical)
    0x05, 0x09,                    //     USAGE_PAGE (Button)
    0x19, 0x01,                    //     USAGE_MINIMUM (Button 1)
    0x29, 0x10,                    //     USAGE_MAXIMUM (Button 16)
    0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
    0x25, 0x01,                    //     LOGICAL_MAXIMUM (1)
    0x95, 0x10,                    //     REPORT_COUNT (16)
    0x75, 0x01,                    //     REPORT_SIZE (1)
    0x81, 0x02,                    //     INPUT (Data,Var,Abs)
    0x05, 0x01,                    //     USAGE_PAGE (Generic Desktop)
    0x09, 0x30,                    //     USAGE (X)
    0x09, 0x31,                    //     USAGE (Y)
    0x09, 0x32,                    //     USAGE (Z)
    0x09, 0x33,                    //     USAGE (Rx)
    0x15, 0x81,                    //     LOGICAL_MINIMUM (-127)
    0x25, 0x7f,                    //     LOGICAL_MAXIMUM (127)
    0x75, 0x08,                    //     REPORT_SIZE (8)
    0x95, 0x04,                    //     REPORT_COUNT (4)
    0x81, 0x02,                    //     INPUT (Data,Var,Abs)
    0xc0,                          //     END_COLLECTION
    0xc0                           // END_COLLECTION
};
```
Error:
```
This device cannot run. (Code 10)

No top level collections were found in this device.
```
My target is to make a Gamepad device with 16 buttons and 2 axis joysticks, i tryied a http://ww1.microchip.com/downloads/cn/AppNotes/cn_91054c.pdf too but this not work too :/ Any ideas/tips?

Thanks

Sebring, 2016-02-22:
Am I missing something, but why not just use this library? - it works as a charm.

MHeironimus, 2016-02-23:
@GAMELASTER
Looks like I misunderstood what you were asking. If you are wanting to create a simpler version of the ArduinoJoystickLibrary, I would suggest looking at the Joystick2 version (since it only has an x and y axis and 16 buttons).

GAMELASTER, 2016-02-23:
@Sebring
I need to simulate a GamePad, so 16 buttons and two axis joysticks and i dont need to use some other joystick things.
@MHeironimus
I want to make custom library for GamePad HID, I did it so similar as this library, but it's not working. I replaced a hid descriptor in Joystick library to my descriptor, and there is working. O_o . Later I will send my code.

GAMELASTER, 2016-02-23:
And one more question, it's exists a tools for debugging HID descriptors or something for checking where is problem with HID?

MHeironimus, 2016-02-23:
The one tool I am aware of is the [HID Descriptor Tool](http://www.usb.org/developers/hidpage#HID Descriptor Tool). This tool allows you to create, edit and validate HID Report Descriptors. It can be downloaded from the following link: [http://www.usb.org/developers/hidpage#HID Descriptor Tool](http://www.usb.org/developers/hidpage#HID Descriptor Tool).

Sebring, 2016-02-23:
I currently use this library for one axis only gamepad. A 10k pot used as a paddle controller for a pong html game (simulating html gamepad), not even using 1 button. So again I recommend you to consider using this library, instead of reinventing the wheel.

GAMELASTER, 2016-02-24:
Ah, please, give me a award for blindness.. I forgoten add to a HID descriptor variable a PROGMEM attribute. For this thing Windows still have problem with a decriptor.

Thanks for help!

closed 2016-02-24 by GAMELASTER