MHeironimus/ArduinoJoystickLibrary #4: Joystick2 and Joystick3 libraries only appear as single joystick

jeqbrown, 2016-03-21:
Hello,

When I use the Joystick2 library, it only appears as a single joystick with 2-axis and 32-buttons. The buttons assigned on the second joystick (Joystick[1]) show as the first joystick (Joystick[0]) button+16... ie- when pressing J2 btn 0 it sends as J1 btn16, J2 btn 4 sends J1 btn20.., etc.. The Joystick3 library does not register ANY joysticks.

I am using Ubuntu 15.04 and jstest-gtk for testing.

MHeironimus, 2016-03-22:
Take a look at this article and see if it addresses your issue:

http://mheironimus.blogspot.com/2015/09/linux-support-for-arduino-leonardo.html

https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/4

jeqbrown, 2016-03-22:
Hi Matthew,

Thank you for taking the time to respond. That is exactly what I was looking for! Also, thank you for your work on the Arduino joystick libraries. The library I the perfect solution for my project. It works great in combination with the SNESpaduino library to enable my now "RPi brained" SNES to work with the original controllers. You are the man I send my appreciation for your hard work.

Thank You,

Jerimiah

MHeironimus, 2016-03-22:
Glad to hear that fixed your problem and that you have found the library useful.

Matt

---
https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/4#issuecomment-199894788

closed 2016-04-03 by MHeironimus