MHeironimus/ArduinoJoystickLibrary #5: Error

GeorgeNidri, 2016-04-19:
Hi, 
when I import the joystick library, I get the following error:

Arduino: 1.7.9 (Windows 8.1), Board: "Arduino Leonardo"

In file included from sketch_apr19b.ino:1:0:

C:\Program Files (x86)\Arduino\libraries\Joystick/Joystick.h:24:17: fatal error: HID.h: No such file or directory

 #include "HID.h"

                 ^

compilation terminated.

Error compiling.

I get that even if I only import the library. Any ideas of why that happens?

Thanks!

MHeironimus, 2016-04-19:
I have tested this on Arduino IDE 1.6.6, 1.6.7, and 1.6.8. I am not familiar with Arduino: 1.7.9.

GeorgeNidri, 2016-04-19:
so should I downgrade?

GeorgeNidri, 2016-04-19:
Yup, it worked. Thanks!

closed 2016-04-21 by MHeironimus

j-keiter, 2018-11-24:
What worked??  The downgrade?  If so what version did you downgrade to.
Thx Jason