MHeironimus/ArduinoJoystickLibrary #6: Events not detected under SDL2

sdumetz, 2016-04-21:
I have ran into it multiple times with this library (or previously with my home made equivalent).

SDL2 does detect a SDL_JOYDEVICEADDED but then no further SDL_JOYBUTTONDOWN is emitted.

It's reproductible with a minimal Arduino code pressing buttons and the following SDL code : 
```
#include <SDL2/SDL.h>
#include <iostream>

int main(int argc, char **argv)
{
    // Notre fenêtre
    bool end = false;
    SDL_Joystick *joy;
    SDL_Event evenements;
    bool terminer(false);

    if(SDL_Init(SDL_INIT_VIDEO |SDL_INIT_JOYSTICK) < 0)
    {
        std::cout << "Error during SDL init : " << SDL_GetError() << std::endl;
        SDL_Quit();

        return -1;
    }
    SDL_JoystickEventState(SDL_ENABLE);
    for(int i=0;i<SDL_NumJoysticks();i++){
      std::cout<<"Joystick N°"<<i<<" : "<< SDL_JoystickNameForIndex(i) << std::endl;
    }
    while(!end){
      while(SDL_PollEvent(&evenements)) 
      {
        switch (evenements.type){
          case SDL_QUIT:
            /* Quit the application */
            std::cout<<"EXIT"<<std::endl;
            end = true;
            break;
          case SDL_CONTROLLERDEVICEADDED:
            std::cout<<"Controller Added"<<std::endl;
            break;
          case SDL_JOYDEVICEADDED:
            std::cout<<"Joystick Added"<<std::endl;
            break;
          case SDL_JOYBUTTONDOWN:
            std::cout<<"button "<<evenements.jbutton.button<< "Pressed"<<std::endl;
            break;
          default:
            std::cout<<"event of type:"<<evenements.type<<std::endl;
        }

      }
    }
    // On quitte la SDL
    SDL_Quit();

    return 0;
}
```

Compile with : `g++ -lSDL2 SDL_test.cpp -o test`

I've tested with debian 8.2 (SDL2 v2.0.2) and arduino Leonardo. I lack the hardware to do proper testing on other platforms / hardware.

I think it might just be a problem in SDL and I'm ready to dig into it but beforehand I'd like to be sure it's a global problem which affect "everyone".

nenad-jalsovec, 2020-11-23:
Have you ever managed to get to the bottom of this?
I ran into exactly the same issue working on a small SDL project on windows with VC++ 2017 with SDL 2.0.9. Same thing with latest SDL 2.0.12.

Joystick was initialized by the book, all other joystick events are firing properly except SDL_JOYBUTTONDOWN.
Can't find any solution to this around the web. Please advise if you found a fix.
Cheers!

mcrfinch, 2021-11-29:
So a year on and I too have discovered an issue with the buttons, I'll leave it here incase its related for or helps someone else.

No button changes detected when using SDL_JoystickGetButton. Work around is to poll events as shown below (QT using VS2019 compiler).

`#define MAXBUTTON 32

bool buttonVal [MAXBUTTON] = {};

void pollButton(SDL_Joystick * controller)
{
    //Event handler
    SDL_Event e;

    while( SDL_PollEvent( &e ) != 0 )
    {

        if(e.type == SDL_JOYBUTTONDOWN)
        {
            buttonVal[e.cbutton.button] = true;
        }
        else if(e.type == SDL_JOYBUTTONUP)
        {
            buttonVal[e.cbutton.button] = false;
        }
    }
    /*
    for(uint8_t i = 0; i < MAXBUTTON; i++)
    {
        if(SDL_JoystickGetButton(controller,i) != 0)
        {
            buttonVal[0] = true;
        }
        else
        {
            buttonVal[0] = false;
        }
    }
    */
}`

ShikyC, 2024-01-04:
I'm working on a personal project and also encountered this issue. After some preliminary testing I can confirm that only `SDL_JOYDEVICEADDED` and `SDL_JOYDEVICEREMOVED` events are detected by SDL2, while all the other inputs like `SDL_JOYAXISMOTION` and `SDL_JOYBUTTONDOWN` are not captured.