MHeironimus/ArduinoJoystickLibrary #8: Various questions...

snagov, 2016-07-10:
- can i configure more than 32 buttons?, changed in joystick.h to 64 buttons, but only showed 32 in joystick properties

- In Joystick properties, not show "Rudder", "Throttle", "Brakes" and "Steering"

Thanks....

MHeironimus, 2016-07-11:
> can i configure more than 32 buttons?, changed in joystick.h to 64 buttons, but only showed 32 in joystick properties

I know the Test dialog in Windows will only show 32 buttons and the `DIJOYSTATE` structure returned by the `IDirectInputDevice8::GetDeviceState` method only supports 32 buttons (see [https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dijoystate.aspx](https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dijoystate.aspx)). The `DIJOYSTATE2` structure supports up to 128 buttons, however (see [https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dijoystate2.aspx](https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dijoystate2.aspx)). I suspect the ability to support over 32 buttons depends on the software that is reading the joystick's value on the host computer. Version 2.0 of the Arduino Joystick Library should support over 32 buttons if the host computer supports it.

> In Joystick properties, not show "Rudder", "Throttle", "Brakes" and "Steering"

The Windows Game Controller Test dialog will only show the first 5 analog inputs after the two that show up in the X/Y Axis area. The additional controls should be there, they just do not show up in the test dialog.

I have been looking for an application that would display more details than the Windows built-in test dialog, but have not found one I like yet. I may have to write my own. If anyone knows of a good program for testing Joysticks/Game Controllers, I would be interested to hear about it.

snagov, 2016-07-11:
Ok...thanks.....8 axies and 32 buttons its ok!

leonardosalvatore, 2016-07-11:
> If anyone knows of a good program for testing Joysticks/Game Controllers, I would be interested to hear about it.

Good job MHeironimus!
Profitably use your software on some Arduino Nano and testing it (a part from games) with:
Linux:
http://pingus.seul.org/~grumbel/jstest-gtk/ 
Windows:
http://www.planetpointy.co.uk/joystick-test-application/

snagov, 2016-07-13:
Hello again....

I found this code that can read 32 switches, but not as to how it works ..... any help?

`byte switchPin[7] = { 4, 5, 6, 7, 8, 9, 10 };

byte switchValue[42];

void setup() {
 for (byte i = 0; i <= 7; i++) {
   pinMode(switchPin[i], INPUT_PULLUP);
 }
}

void loop() {
 byte n = 0;
 for (byte i = 0; i <= 7; i++) {
   pinMode(switchPin[i], OUTPUT);
   digitalWrite(switchPin[i], LOW);
   for (byte j = 0; j <= 7; j++) {
     if (j != i) {
       switchValue[n++] = digitalRead(switchPin[j]);
     }
    }
   pinMode(switchPin[i], INPUT_PULLUP);
 }
}`

I thing, i can use this wiring:

[http://image.prntscr.com/image/2fd75e6eaf0a4cabb1df017851119998.jpeg](url)

MHeironimus, 2016-08-19:
@leonardosalvatore Thanks for the joystick testing application links. They were helpful.

leonardosalvatore, 2016-08-19:
You are welcome, I'm using your library for a DIY project and others which
are coming...
https://www.youtube.com/watch?v=CGd8vZrIq1Y

https://www.youtube.com/watch?v=SljBrh-aOS4

MHeironimus, 2016-08-19:
@leonardosalvatore Thanks for sharing those videos. I always enjoy seeing how the library is being used.

closed 2016-09-06 by MHeironimus