MHeironimus/ArduinoJoystickLibrary #9: Feature request: Force Feedback/Rumble/Vibration support

jfrmilner, 2016-07-12:
I’d like to request support for Force Feedback/Rumble/Vibration (controller motor control). I’ve seen that other albeit old ATmega32U4 projects have implemented this, for example
https://github.com/tloimu/adapt-ffb-joy/blob/wiki/USBFfbHid.md
Ideally this would appear as a joystick with advanced force feedback features in a Windows machine without need for installing any device drivers like the above project.
Assuming this is possible I think it would be a welcome addition to an already great library, what do you think?

MHeironimus, 2016-07-27:
I like the idea. I will keep this issue open until that feature is added or it is determined that this cannot be done.

zaped212, 2016-09-22:
Just a heads up, I was able to use the single joystick library, keyboard, and Serial over the usb connection of a leonardo. I planned on using the Serial connection as a controll channel and could be used for feedback depending on your application.

MHeironimus, 2016-09-22:
@zaped212 That sounds like a good option. Let us know how it goes or if there are any changes to the Joystick Library that would make this easier.

zaped212, 2016-09-22:
The serial control channel only works in custom applications so a native option would still be best ( work in existing games that would utilize it ). I was able to use the serial connection with no modifications but unfortunately my project has moved me away from that path.

Was going to use the control channel to enable/disable buttons&LEDs per different configurations but never got around to fully implementing the control data.

baritonomarchetto, 2016-12-15:
The Force Feedback/Rumble/Vibration  would be a great addition to an awesome library, I agree 100%!

dchote, 2017-09-19:
I'd love to see this feature too!

gagagu, 2017-09-28:
Force Feedback would be great!

gagagu, 2017-10-24:
I've found some useful information about Force Feedback HID on internet. I'm not deep enough in USB HID programming but may be someone can use this informations?
http://www.microchip.com/forums/m320225-p2.aspx

Adrianotiger, 2018-02-28:
Maybe this links can help too (from the vJoy emulator):

HID Descriptor:
https://github.com/shauleiz/vJoy/blob/master/driver/sys/vjoy.h

Reference:
http://forums.obdev.at/viewtopic.php?f=8&t=4313#p14541

But I am not able to test it in the vjoy and can't see this functionality on Windows, even if it should be  implemented.

uzairali001, 2019-02-21:
So does anyone able to make FF enabled device yet?

Alacritous, 2019-07-30:
I was just looking into this, and I too would like to see this implemented.  I don't have the skills to do it myself, but here's the usb.org specifications for physical interface devices over USB. (Force feedback)

https://www.usb.org/sites/default/files/documents/pid1_01.pdf

hoantv, 2019-09-19:
Joystick is great. Waiting for ffb support.

MC42, 2019-11-08:
Any word on this potentially ever coming?  I have a project I'd love to use it for but it doesn't look like there's any work for it...

Nikhil707, 2020-04-19:
Anybody got success??here, i got an russian link which you can translate and read, seems like that guy has reverse engineered some logitech steering  wheel and since my programming skills are poor so cant understand much from this,hope some of you will... thanks in advance

[https://cxem.net/mc/mc207.php](url)

hoantv, 2020-04-22:
> Anybody got success??here, i got an russian link which you can translate and read, seems like that guy has reverse engineered some logitech steering wheel and since my programming skills are poor so [...]

You can check my repo.

Nikhil707, 2020-04-30:
> You can check my repo.

Thank you so much!!

YukMingLaw, 2020-05-11:
I have build up a library that support ffb,based on MHeironimus and hoantv's work.check out ->https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary

hoantv, 2020-05-11:
> I have build up a library that support ffb,based on MHeironimus and hoantv's work.check out ->https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary
Good job. Why dont you add ffb calcultor?

hmx7, 2021-06-24:
@MHeironimus 力反馈/隆隆声/振动真是一个很不错的主意，我建议也增加按键状态反馈功能。

Skuzee, 2022-06-04:
Do windows HID compliant game controllers support a rumble status by default? It would be nice to define a controller that has a rumble feature so I could bind it in Dolphin. (Gamecube/wii emulator) Just as we use `Joystick.setXAxis` a `Joystick.getRumble` would be nice. 

Currently using an atmega32u4 to read a gamecube controller, and I can set the rumble of the gc controller easily, I just need a way for Windows to communicate the rumble status to the Arduino.

Thanks!