MHeironimus/ArduinoJoystickLibrary #10: hid.h

djdarlo, 2016-07-26:
I've copied the library file as instructed but getting an error saying the hid.h file cannot be found.  Any ideas?

MHeironimus, 2016-07-26:
Can you indicate which Arduino board you are targeting, what version of the Arduino IDE you are using, which version of the ArduinoJoystickLibrary you are using, and a copy of the error message?

Thanks,
Matt

djdarlo, 2016-07-27:
Hi Matt.  I am using a Leonardo.  IDE is 1.7.10 and the error message I am
getting is

Arduino: 1.7.10 (Windows 7), Board: "Arduino Leonardo"

In file included from JoystickButton.ino:13:0:

C:\Users\david.j.darlington\Documents\Arduino\libraries\Joystick/Joystick.h:24:17:
fatal error: HID.h: No such file or directory

 #include "HID.h"

                 ^

compilation terminated.

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Thanks in advance,
David

MHeironimus, 2016-07-27:
I am not familiar with version 1.7.10 of the Arduino IDE. The latest version I am aware of is version 1.6.10 (https://www.arduino.cc/en/Main/Software). Try using that version and see if it resolves your issue.

djdarlo, 2016-07-27:
Many thanks.  This now compiles perfectly.

D

closed 2016-07-27 by MHeironimus