MHeironimus/ArduinoJoystickLibrary #11: Incompatibility with Keyboard.h

LeoDJ, 2016-07-27:
I have not tested much and don't really know what is causing it.
I wanted my Leonardo to be both a joystick and a keyboard. 
As soon as I added the `#include <Keyboard.h>` my mouse cursor was controlled by the values I was sending to the X and Y axes of the Joystick.
I don't even have to call `Keyboard.begin()`.

My guess is that some HID-magic is going wrong between the two libraries, so the values get sent as mouse axes instead of joystick axes.

Probably reproducible by running the following code:
```C++
#include <Keyboard.h>
#include <Joystick.h>

void setup() {
  Joystick.begin();
}

void loop() { //should move joystick stick in a square really fast
  Joystick.setXAxis(-5);
  Joystick.setYAxis(-5);
  Joystick.setXAxis(5);
  Joystick.setYAxis(5); 
}
```

MHeironimus, 2016-07-28:
What OS are you using (e.g. Windows, Linux, Mac, etc.)? If it is Linux, I know some Linux distributions include a driver that causes the joystick to be a virtual mouse (see https://ubuntuforums.org/showthread.php?t=1879616, https://bbs.archlinux.org/viewtopic.php?id=119421, https://bugs.launchpad.net/ubuntu/+source/xorg/+bug/277915, etc.).

MHeironimus, 2016-08-22:
Are you still having this issue?

Tarnadas, 2016-09-05:
I have the same issue with the Mouse.h library. As soon as I include both Mouse.h and Joystick.h, calling Mouse.move() with any arbitrary parameter will let my cursor jump to the bottom right corner of my screen.
My board is a Genuino Micro and my OS is Windows 7 x86_64 processor.

I just tested it under Android 5.1.1 with ARMv8 and it doesn't seem to jump. Will make a few more tests

Nefando, 2017-10-18:
Hi,
I had a slightly different issue on using Keyboard.h library along with Joystick.h, using 2 buttons (one as joystick button and the other as keyboard stroke).
While the joystick button input was correctly read in the gaming devices property window and the keyboard stroke worked properly in Notepad, I encountered a strange double input while using the keyboard button inside the game control configuration (specifically I tried it in Arma 3): when asked to press the button to link to desired control, pressing the keyboard stroke button gives the correct keyboard stroke, but also gives the input of the joystick device button. 
Here is my code (is a work in progress, please don't mind the sintax and comments as I'm new to Arduino :) ):

`#include "Joystick.h"

#include <Keyboard.h>

Joystick_ Joystick;

// Set analog axis pin
const int X_PIN = A2;
const int Y_PIN = A3;
const int Xr_PIN = A4;
const int Yr_PIN = A5;

// Stick values
int xValue, yValue, yValueRev, xrValue, yrValue;

void setup() {

  // Initialize Button Pins
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);

  // Initialize analog stick.
  pinMode(X_PIN, INPUT);
  pinMode(Y_PIN, INPUT);
  pinMode(Xr_PIN, INPUT);
  pinMode(Yr_PIN, INPUT);

  // Initialize Joystick Library
  Joystick.begin();

  // Initialize Keyboard Library
  Keyboard.begin();
  
  //Serial.begin(9600);
}

void loop() {

  // Read pin values
  for (int index = 0; index < 2; index++)
  {
    int currentButtonState = !digitalRead(index + 2); 
      switch (index) {
        case 0: // BUTTON 1
          if (currentButtonState == 1) {
          //Joystick.setButton(0, currentButtonState);
          Keyboard.write(65);
          delay (50); // Delay for avoiding char repeat
          }
        case 1: // BUTTON 2
          Joystick.setButton(1, currentButtonState);
          break;        
      }
  }
  
  // Read analog stick position.
  xValue = analogRead(X_PIN);
  delay (5);
  yValue = analogRead(Y_PIN);
  delay (5);
  yValueRev = map(yValue, 0, 1023, 1023, 0);
  xrValue = analogRead(Xr_PIN);
  delay (5);
  yrValue = analogRead(Yr_PIN);
  delay (5);

  // Set the actual joystick position
  Joystick.setXAxis(xValue);
  Joystick.setYAxis(yValueRev);
  Joystick.setRxAxis(xrValue);
  Joystick.setRyAxis(yrValue);

  //Serial.print ("Asse X: ");
  //Serial.print (xValue);
  
  //Serial.print (", Asse Y: ");
  //Serial.println (yValueRev);

  delay(50);
}`

I'm on Windows 10 64bit, Arduino 1.8.4 and using Arduino Micro board.

PS: sorry for my English, not my native language

JayToltTech, 2022-03-25:
I’d suggest using ‘bisect debugging’ to try and figure out the problem.  It looks like this:

* Include keyboard.h in your project
* Verify failure
* Comment out the top half of keyboard.h
* Does it fail in the same way? If so, comment out the top 1/4.  If not, comment out the bottom half
* Follow this until you’ve isolated the code that causes the failure

Obviously it’s not quite this simple, as the code needs to compile, so you have to choose intelligent split points.  But you get the idea.

If finding split points is really hard, you can invert the process.  Include keyboard.h, but comment out all the lines of code.  Add in code sections until the failure reoccurs.

ATMakersBill, 2022-03-25:
Ron showed me this issue earlier today.  This is a problem with the XBox Adaptive Controller's USB stack.  There is no (known) way around it.

The USB stack will not recognize a composite device.  When a composite HID device is presented, the XAC simply refuses to talk to it.  This is a known issue and one that I reported to Microsoft through many channels years ago.  A PDF that details the issue is here: https://github.com/adafruit/circuitpython/files/3005031/XBox.Feedback.pdf

A bug for CircuitPython is also here which has more detail: https://github.com/adafruit/circuitpython/issues/1696

As soon as the keyboard.h and mouse.h are both included, the USB descriptor switches to make a composite device which causes your issues.

Sorry to be the bearer of bad news.

MHeironimus, 2022-03-30:
Closing this issue since it is not an issue with the library, but with the host device.

closed 2022-03-30 by MHeironimus