MHeironimus/ArduinoJoystickLibrary #12: Sample code

djdarlo, 2016-07-29:
I have an arduino leonardo with a joystick (4 buttons) and 6 buttons.  Is there any sample code out there which will make this look like a joystick?  I've tried the test joystick code but it doesn't seem to output anything.  Do I need the joystick (up, down, left, right) outputs plugged to specific pins on the arduino and likewise for the six buttons?

Thanks in advance,
David

MHeironimus, 2016-07-29:
I would suggest installing Version 2.0 (currently in Beta Testing - https://github.com/MHeironimus/ArduinoJoystickLibrary/tree/version-2.0) and take a look at the JoystickButton example.

MHeironimus, 2016-08-22:
Were you able to wire up your Arduino Leonardo to your buttons and get your project working?

djdarlo, 2016-08-22:
I did in the end thanks.  All up and working.  Great fun.

D

closed 2016-09-05 by MHeironimus