MHeironimus/ArduinoJoystickLibrary #13: Newbie question: Using a potentiometer as analog axis?

olliraa, 2016-08-12:
Just found this great library :) Thank you already in advance MHeironimus! I want to build a hand brake for my car sim. So I thought (for the electronics) I only need a suitable version of Arduino, a potentiometer and this library. Is this correct? I tried to find the part of the code that specifies the analog input pin for the potentiometer, but could nhot find it. Am I missing something very elementary here? (probably yes ;)

Thanks again!

sdumetz, 2016-08-16:
Hi,
You can use any pin for your potentiometer. As long as you acquire a position value. Many tutorials on the net will explain to you how to make such a measurement with a potentiometer

This library only need the resulting value, using : 

    Joystick.setXAxis(127*pot_pos/1023);

This will give you a X axis of 0..127 when moving your potentiometer. You can also use Y axis or any other analog joystick part. See [examples](https://github.com/MHeironimus/ArduinoJoystickLibrary/blob/master/Joystick/examples/UsbJoystickTest/UsbJoystickTest.ino) for more.

olliraa, 2016-08-17:
Thank you sdumetz :) I hope the postman would bring me my Leonardo asap :)

closed 2016-08-22 by MHeironimus

MHeironimus, 2016-08-25:
@sdumetz  answer is spot on. Thanks for contributing (it some times takes me a while before I can get to answering these questions).

I hope to include an example of how to do this in Version 2.0 of this library.