MHeironimus/ArduinoJoystickLibrary #14: is it possible to Rename the HID Controller

flow1986, 2016-08-12:
Hi,
i´d like to connect 4 Arduinos to my PC and my Raspberrry.
is it possible to change the HID name?
so that i don´t have:
arduino leonardo
arduino leonardo
arduino leonardo
arduino leonardo

but my own names?

MHeironimus, 2016-09-05:
I have not been able to figure out how to do this yet, but it seems like it should be possible. I am leaving this issue open as a future enhancement. If anyone knows how to do this, let me know.

ytmytm, 2016-09-05:
At one time I tought that recompiling boot loader would be enough:
http://forum.arduino.cc/index.php?topic=249011.0

But apparently not:
http://forum.arduino.cc/index.php?topic=275686.0

However I never tried to do it myself.

marekhalmo, 2016-09-08:
Have you tried to change the string in HID.h or HID.cpp ? I think i saw something like this somewhere in the Arduino IDE base libraries..

trevorirwin, 2016-09-19:
I found some steps for changing the name, courtesy of this YouTube video: https://www.youtube.com/watch?v=nrT7WXFEEZI

1. Go to control panel.
2. Open 'Devices and Printers'.
3. Find your controller , right click ~ properties.
4. Go to tab 'Hardware' , and open properties on HID-compliant...
5. Go to tab 'Details' and select on 'Property' ~ Hardware Ids.
6. Remember your value , VID_045E&PID_02A1 " Every controller has different value. "
7. Go to start , and open regedit.exe.
8. Go to :
A. HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_045E&PID_02A1
B. HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_045E&PID_02A1
9. Change OEM Name to your preferences.
10. Done!

However, this changes the name for ALL "Arduino Leonardo" devices. I'm not sure how to change the name for just one specifically. The joystick devices seem to identify to windows based on VID and PID. If there were some way to safely change these, it might be possible to name them individually.

Based on some of my previous experiments, extreme care must be taken here, however. If you change the VID or PID of the hardware Leonardo, you run the risk of the Arduino IDE not recognizing it for sketch uploads anymore. There might be a way to modify it for just the joystick device, but I don't know enough about HID to say. Hopefully @MHeironimus or someone else with more talent can use this as a starting point.

bjoernboeckle, 2016-11-29:
Unfortunately I couldn’t figure out how it can be influenced from within the sketch.
But you can change the name globally by modifying the “boards.txt” file.
This will modify the name for all devices programmed. So if you want to have Name A, B and C for your devices you need to change the “boards.txt” to “A”, program the A device, then change the boards.txt to “B” then program the B device etc.
The boards.txt can be found in the Arduino installation folder, for instance:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\boards.txt

In case you’re device is reported as "Arduino Micro", then you need to change the “Arduino Micro” entry to whatever you want (in case its an Arduino Leonardo then change the appropriate usb_product entry), for instance for a Arduino Micro change:
micro.build.usb_product="Arduino Micro" 
to
micro.build.usb_product="LuckyB's Joystick"
Result:

[image]

MHeironimus, 2016-11-29:
@bjoernboeckle Thank you for posting your findings.

maxx8888, 2017-01-01:
Hi,

Thanks for the Tip with the Product Name change. It was working perfectly for me also. However, the 3 Joysticks itself are still named Arduino Leonardo. Please refer to attached Pictures.

Therefore i'm still struggeling within my Flight Simulator to find correct Joystick, as i have connected 3 Leonardo Boards with 3 Joysticks each. Within Game, they are all named "Arduino Leonardo" :-)

[image]

[image]

Dakor91, 2017-03-04:
From what i have found, you have to modify the bootloader and recompile it, to change the displayed name.

You are legally not allowed to change the name and use Arduino LLCs USB Vendor Id, though. This would mean you have to buy a Vendor Id yourself, which costs $5.000: http://www.usb.org/developers/vendor/

I don't think it's an issue if you don't want to sell / release your product, though.

MHeironimus, 2017-03-05:
@Dakor91 - Thank you for posting your findings.

silveirago, 2017-07-17:
@Dakor91 Thank you very much for that info! It solved my problem.

trevorirwin, 2017-08-01:
@Dakor91 Thanks for that! While the solution I posted earlier worked for renaming a single class of device, your solution is so much better because it allows my devices to identify differently to a single PC.

JamesCoyle, 2017-09-28:
When I do as @Dakor91 suggested it shows up with the new name in the devices page in the new windows 10 settings but no longer gets recognized as a game controller for some reason... Any ideas?

condac, 2018-03-31:
If I change the VID and PID it does not show up in Game Controllers in windows. It Shows up as a HID-compliant device and USB Input Device in device manager and under other devices it shows up as a serial device with no drivers. If I force the arduino drivers on the serial device it will work as a serial device but it still doesn't show up in Game Controllers.  Any ideas how to fix this? I can have 2 different names if I change the PID to leonardos PID on one and the micro on another, but if i want more unique names changing the PID to a random value would be nice.

condac, 2018-03-31:
There is a bug somehow. I uploaded an example to a new adruino with new random VID and PID and it works. So if i change the VID and upload my code first, my code does not work. But when I load it with an example and plug it in then i can reload it with my code and it works.  Some how having this joystick initializer does not work ( the example for driving controller):
`Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS, 4, 0,
  false, false, false, false, false, false,
  false, false, true, true, true);`
But if i change it to the example for gamepad it works fine. Or if i have used the gamepad example before on that VID it starts to work.

tseiracq, 2019-06-11:
Hello,
Sorry for the up but did anyone find a way to change the HID name of a single (or two) controller on a same Arduino ?
I'm currently working on a project and that would be very helpful if there was a way to use only one board instead of two

alijani1, 2020-05-17:
> Hello,
> Sorry for the up but did anyone find a way to change the HID name of a single (or two) controller on a same Arduino ?
> I'm currently working on a project and that would be very helpful if there was a way to use only one board instead of two

I've been searching for quite a long time and not found any way to change individual controllers under the same device :(   you can change the name of the HID with borads.txt but individual controllers will all have same name you used.

phil123456, 2021-01-20:
it does not work at all

MHeironimus, 2021-01-24:
There is a demonstration on how to do this towards the end of this video: [Arduino Joystick 2.0 Library - Beginners Guide](https://www.youtube.com/watch?v=hoCOq9Ngp44).

jormc, 2021-03-05:
Thanks for the info, I was looking for it :-)
I'll try!

Shchava, 2021-12-25:
> There is a demonstration on how to do this towards the end of this video: [Arduino Joystick 2.0 Library - Beginners Guide](https://www.youtube.com/watch?v=hoCOq9Ngp44).

Thanks for the tip,
I wasn't able to add additional avr boards.txt file as described in video, but i achived goal by appending new board config with tweaks described in video to the end of default (placed AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4) boards.txt file.
So now my config shows up in board list, and after dlash it changes HID name 
[image]
[image]

racdavies, 2021-12-29:
Hi, ive been having this very same naming issue for....months, i have 5 button boxes for my sim rig which i bought from an ebay seller , they all use Arduino Leonardo boards....or so i thought, after opening up one of the boxes i found that they are Pro Micro boards and not official Arduino.

After many attempts at trying different things... like using the Musinou board txt file i found out that as well as using the board txt files you also need to upload a a sketch at the same time, in my case i used a 32-FUNCTION-BUTTON-BOX-master sketch from AMSTUDIO.

Although i had several hurdles to get over... getting the correct keyboard library, finding the correct location of the board txt file for editing....Windows11 seems to block access to the Arduino IDE app location....nice one MS and much more fiddling.

So now everything is in place, i change the leonardo.build.usb_product=  name to my chosen name and the sketch i acquired ....upload and voila, shows up in device and printers under controllers but after clicking properties  it still showed as Arduino leonardo, i tested the board running Asseto corsa and BAM....Leonardo device detected.

Now for the fix, the vid and PID codes have to be changed from the original but also they must not clash with any other usb devices, unless you have a complete list of your device id`s....this seems to be pot luck, I have not tested the same new code with my other boxes but as precaution im using different code for each box, it means editing the board txt file 5 times but it does mean my games detect different devices instead of multiple Arduino Leonardo`s. 

Licensing issues, all my boards are Arduino compatibles made by SparkFun, they state on there website that the user is free to change the name of there boards and does not require a license to do so.

I have tested this method with Asseto corsa and American truck sim and both games detect my board as the name given.

This process can be hit and miss, i suggest testing on a board thats new and unwired to a device, In my endeavor... I broke one button box by accidentally breaking off the usb socket from the Pro Micro and killed another by trying to jump the onboard reset/gnd after a bad upload.

Hyratel, 2022-03-24:
you can add several defines to a new submenu https://gist.github.com/Hyratel/80017369fedd1bbc9eef4c8e7a896225

racdavies, 2022-03-24:
> Hi, ive been having this very same naming issue for....months, i have 5 button boxes for my sim rig which i bought from an ebay seller , they all use Arduino Leonardo boards....or so i thought, [...]

Hi, I actually managed to sort it out shortly after posting here, my main issue was trying to update using IDE without adding a sketch, the guy i bought the boxes off sent me new sketch's for all 5 boxes as they all have different matrix`s, now all i need is about an extra 5 hours a day to have time to actually use my sims.... either that or win the lottery lol

vospascal, 2022-05-17:
@MHeironimus  https://github.com/abratchik/HIDPowerDevice
https://github.com/arduino/ArduinoCore-avr/pull/387 

this might be something we can work with :) i think it would be realy awesome to support custom hid names

Hyratel, 2022-05-17:
@MHeironimus https://github.com/Hyratel/LeonardoPIDboards have a looky here. I went through and spun this up. you ***have*** to use different PIDs because windows Caches the name in the registry. (the 0x5000 range on PID.codes is fairly empty so I feel fairly safe using it. it's not... great because it collides with 1 or 2 already listed, but at the time I just did not have the patience to look for 50 consecutive unused values)

Hyratel, 2022-05-17:
@vospascal no, it loads the names into the USB stack on the arduino, so it will announce itself with the name you set. it will work on any machine that way. the Cacheing is when windows recieves a USB announce, it loads that into the registry and any changes in device name will *not* be reflected because it's pulling the names from cache not device

vospascal, 2022-05-18:
yea i know the methode but its not realy same as setting names vendor in the hid report

Clemy01, 2022-11-19:
Hello,
Any help for doing this for an Arduino DUE?
It works for my Micro Pro (F14 HCU), but not my Arduino DUE (F14 Button Box).
Thanks a lot!

(Screen : https://media.discordapp.net/attachments/349553308949544964/1043537031843815454/image.png )

mwwhited, 2023-03-14:
> You are legally not allowed to change the name and use Arduino LLCs USB Vendor Id,. So you have to buy a Vendor Id yourself, which costs $5.000:

There is nothing about legality of using your own unregistered vendor ids.  It could annoy the USB committee but there is no crime they can charge you with for using an unregistered vendor ID.  Though keep in mind you can falsely trigger different drivers on your computer if you have a conflict but there is no legal or civil actions against you if you choose to change these values.  They could get you for trademark/copyright if you use their logos and trademarks without paying a license... but if you  take an off the self USB host device and change the ID values for your own convenience they can't do anything about it.

mwwhited, 2023-03-14:
BTW, many suggest you use the vendor id 0xf055 mainly because it looks like FOSS but partially because it annoys the USB-IF and a few other vendor groups.

Amy-81, 2024-12-28:
Hi, I struggled with this issue myself, so I decided to create step-by-step guide. It works for me on Win 10. This guide is based on below video.

https://www.youtube.com/watch?v=YAbi_AqF7aQ&list=PLqIYovI4NEIfRrr-byoedK3_iY2MHQ8XD&index=1&t=702s

He starts talking about HID Controller names issue at 7:10 and more about how to do it starts at 9:00.

which is similar to this one (starts at 19:43)

https://www.youtube.com/watch?v=hoCOq9Ngp44

1. close Arduino IDE
2. go to "C:\Users\ 'your user name' \AppData\Local\Arduino15\packages" (NOTE that "AppData" folder is hidden, so you need to change Windows settings to show hidden and system folders)
3. create new folder "custom" (or any name you like, this name will show up in Arduino IDE in "Tools>Boards>" menu)
4. inside "custom" folder create new folder "hardware"
5. go to https://liveelectronics.musinou.net/MIDIdeviceName.php scroll down (around 1/3 from top of website) and click "download this"
6. unpack .zip file and you'll see "musinou" folder. From this folder copy "avr" folder to "hardware" from step 4
7. enter "avr" folder (the one in C:\Users\ 'your user name' \AppData\Local\Arduino15\packages\custom\hardware\ ) and open "Boards.txt" file
8. you need to find lines like below and change "vid=" and "pid=" to other number (e.g. increase them by 1) and we'll see if that works with other USB devices connected to your comp.

for Arduino Leonardo (or clones):

leonardomusinou.name=Arduino Leonardo musinou

leonardomusinou.build.vid=0x2342
leonardomusinou.build.pid=0x8032
leonardomusinou.build.usb_product="Arduino Leonardo musinouu"

or for Arduino Micro (or clones such as Pro Micro)

micromusinou.name=Arduino Micro musinou

micromusinou.build.vid=0x2340
micromusinou.build.pid=0x8030
micromusinou.build.usb_product="Arduino Micro musinou"

9. change "Arduino Leonardo musinou" or "Arduino Micro musinou" to the name you'd like for your board to be (if you use Arduino Leonardo board (or clone) you need to change info for Leonardo. If you use Arduino Micro board (or clone such as Pro Micro) you need to change info for Micro)
10. save "Boards.txt" file
11. restart Arduino IDE
12. select your new board from "Tools>Boards>custom-avr (in Sketchbook)>" menu
13. upload your sketch to MCU. From now on it should be visible to Windows under new name you gave in step 9
14. press Win+R on your keyboard, type joy.cpl and press Enter. Game Controllers window will open, where you can see your Arduino game controllers with custom names (see Note 4 below)

Note 1: You can create folder "hardware" in "Documents>Arduino>" folder as well, instead of "C:\Users\ 'your user name' \AppData\Local\Arduino15\packages" like in video from link above. It works either way...

Note 2: Inside "Boards.txt" file you can create multiple boards to use in Arduino IDE. Just copy and paste all lines for the board you need (Leonardo or Micro) and remember to give them unique "name=", "vid=", "pid=" and "usb_product=".

Note 3: You can change "leonardomusinou" or "micromusinou" to any name you like. Just make sure name is same for all lines of MCU you use.

Note 4: If for some reason your custom game controller doesn't show up, go to your sketch in Arduino IDE and in constructor change JOYSTICK_TYPE_MULTI_AXIS to JOYSTICK_TYPE_JOYSTICK. This trick worked for me.

Hope that will help and save some time when building custom game controllers.

FAO creator of Joystick library. Please feel free to add this guide to your library, so new library users won't need to search online for this.

Hyratel, 2025-01-02:
or you could sidestep ALL OF THAT and try https://github.com/Hyratel/LeonardoPIDboards

it adds a new Board Option under my-boards_avr and has a large list of presets that is easily extended

Amy-81, 2025-01-03:
Great if you need copy-paste solution without need to learn anything and if you use Leonardo board only, which I don't. Using my guide above you can have custom Leonardo or (Pro) Micro boards with any name you like. Also, if there is clash with other HID on your system, you know what to change to fix it.

Amy