MHeironimus/ArduinoJoystickLibrary #15: How can I increase the axis range to 1024

Skyou82, 2016-08-22:
That's great job man! Thank you very much!

I need 1024 resolution or more for XYZ axis, I try to change descriptor report and use 16bit value, but that also dosn't work.

    0x15, 0x00,			                //   LOGICAL_MINIMUM (0)
    0x27, 0xFF, 0xFF, 0x00, 0x00,      //   LOGICAL_MAXIMUM (65535)
    0x75, 0x10,			               //   REPORT_SIZE (16) 

and changed 8 bit to 16 bit
    int16_t	 xAxis;
    void Joystick_::setXAxis(int16_t value)   also in Joystick.h and Joystick.cpp

    uint16_t data[JOYSTICK_STATE_SIZE];

There was no feedback in windows joystick setting. What's wrong with my code?

Thanks again!

MHeironimus, 2016-08-22:
Version 2.0 of this library should give you the precision that you need: https://github.com/MHeironimus/ArduinoJoystickLibrary/tree/version-2.0

Let me know if that version works for your project.

Skyou82, 2016-08-22:
Thank you so much!  I will try it!

I just want to know if possible to modify old version to get high resolution axis range?

MHeironimus, 2016-09-05:
In addition to the changes you made above, you will also need to update the size of the X, Y, and Z axis variables in the data array that appears in Joystick_::sendState method. You can refer to the code changes made in Version 2.0 for reference.

closed 2016-09-05 by MHeironimus