MHeironimus/ArduinoJoystickLibrary #16: Android A/B/X/Y button mappings

marekhalmo, 2016-09-05:
Could you add the possibility to press button A/B/X/Y on android gamepad?

The mappings of this joystick are arround 188 (index 0) and higher.. the button mappings for standard android buttons are;
A: 96
B: 97
X: 99
y: 100
RT: 103
LT: 102
TR: 107
TL: 106
Start: 108
Volume: 183
Back: 4

This button mappings are taken from NVIDIA Shield Portable and should be working for most of the Android Gamepads

MHeironimus, 2016-09-06:
Sorry. I am not familiar with Android Gamepads. This joystick library uses the standard USB report descriptors for a PC/MAC/Linux joystick/gamepad. If anyone is familiar with how Android Gamepads work, I would be interested in learning more about them.

marekhalmo, 2016-09-06:
Can you at last point me to a line in code where the number 188 could be generated when pressing a button with index 0? I'm reading the code but not entirely sure if i understand it correctly..

marekhalmo, 2016-09-06:
I was reading the code and understand that you send the button code as data in data[0-3].. but i don't understand how to send a different codes...

MHeironimus, 2016-09-06:
In Version 1.x each button is a bit in the first four bytes of the data array. Button 0 is bit 0 in byte 0, Button 1 is bit 1 in byte 0, Button 8 is bit 0 in byte 1, etc. I never use numbers like 188, so I am not sure where that value is coming from.

Is it possible you have multiple buttons pushed and are reading all of the button values at once? If so 188 could represent 10111100, which would be buttons 7, 5, 4, 3, and 2 (or possibly 11001011 - which would be buttons 7, 6, 3, 1, and 0).

marekhalmo, 2016-09-06:
It is not a multi button command.. value 188 is generated if i send button 0 command only.

Sorry for my long reply but this is my current state of research:
I have been searching for Android HID Descriptors and can't really find any.. the closest thing that i found is for OUYA gamepad (Ouya runs modified Android - Post here: http://forum.xda-developers.com/showthread.php?t=2351437)

The gamepad buttons A/B/X/Y are bound using following entry in the HID Descriptor lines:
 Usage(16)
        Button.0001
        .
        .
        .
        Button.0010
      Logical Minimum(0)
      Logical Maximum(65535)
      Physical Minimum(0)
      Physical Maximum(65535)
      Report Size(1)
      Report Count(16)
      Report Offset(96)
      Flags( Variable Absolute )

Which translates to:
Button.0001 ---> Key.BtnA
Button.0002 ---> Key.BtnB
Button.0003 ---> Key.BtnC
Button.0004 ---> Key.BtnX
Button.0005 ---> Key.BtnY
Button.0006 ---> Key.BtnZ
... e.t.c.

The important part is 
Report Offset(96) which shifts the button reported to the proper region (e.g. Button 0 is reported as 96)

So the solution is to add Report Offset entry to the HID descriptior in your code.. but i can't find the way to do this using HID Descriptor Tool (there is no such entry possible).

I was trying to manually decrypt the OUYA hid descriptor (numbers at the start of the descriptior text in the linked pag).. but so far i had no luck identifying the proper key/value pair in HEX

I'm also reading the USB specs but no luck finding any info there :)

Any ideas how to get the codes for Report Offset(96)?

marekhalmo, 2016-09-06:
I was also thinking about using the keyboard SELECT key.. might work as well.. will see later if this leads to anyting successfull

marekhalmo, 2016-09-07:
Another update on what OUYA HID is descriptor contains...
This is what i was able to decrypt (manually because the tool does not support some of the codes marked by ?????????)

05 01 Usage page Generic Desktop
09 02 Usage MOUSE
a1 01 	COLLECTION (application)
85 06 ???????????????????
09 01 	Usage POINTER
a1 00 		COLLECTION (physical)
05 09 		Usage page Button
19 01 		Usage minimum (button 1)
29 03 		Usage maximum (button 3)
15 00 		Logical Minimum (0)
25 01 ???????????????????
75 01 		Report size (8)
95 03 		Report count (3)
81 02 		INPUT(Data, Var, Abs)
75 05 		Report size (8)
95 01 		Report count (1)
81 01 		INPUT(Const, Arry, Abs)
05 01 Usage page Generic Desktop
09 30 Usage X
09 31 Usage Y
15 81 		Logical Minimum (129 == -128)
25 7f ???????????????????
75 08 		Report size (8)
95 02 		Report count (2)
81 06 		INPUT(Data, Var, Rel)
09 38 Usage Wheel
15 81 		Logical Minimum (129 == -128)
25 7f ???????????????????
75 08 		Report size (8)
95 01 		Report count (1)
81 06		INPUT(Data, Var, Rel)
c0 			END COLLECTION
c0 	  	END COLLECTION
05 01 Usage page Generic Desktop
09 05 Usage (Gaming controls)
a1 01 	Collection (application)
85 07 ???????????????????
a1 00 		Collection (physical)
09 30 			Usage X
09 31 			Usage Y
15 00 			Logical Minimum (0)
26 ff ff 		Logical Maximum(65535)
35 00 ???????????????????
46 ff ff 		Physical Maximum(65535)
95 02 			Report count (2)
75 10 			Report size (16)
81 02 			INPUT(Data, Var, Abs)
c0 	  		END COLLECTION 
a1 00 		Collection (physical)
09 33			Usage Rx
09 34 			Usage Ry
15 00 			Logical Minimum (0)
26 ff ff 		Logical Maximum(65535)
35 00 ???????????????????
46 ff ff 		Physical Maximum(65535)
95 02 			Report count (2)
75 10 			Report size (16)
81 02 			INPUT(Data, Var, Abs)
c0 	  		END COLLECTION  
a1 00 		Collection (physical)
09 32 ???????????????????
15 00 			Logical Minimum (0)
26 ff ff 		Logical Maximum(65535)
35 00 ???????????????????
46 ff ff 		Physical Maximum(65535)
95 01 			Report count (1)
75 10 			Report size (16)
81 02 			INPUT(Data, Var, Abs)
c0 			End collection
a1 00 		Collection (physical)
09 35 			Usage Rz
15 00 			Logical Minimum (0)
26 ff ff 		Logical Maximum(65535)
35 00 ???????????????????
46 ff ff 		Physical Maximum(65535)
95 01 			Report count (1)
75 10       	Report size (16)
81 02 			INPUT(Data, Var, Abs)
c0 			End collection 
05 09	Usage page Button ---------------------------------------
19 01 	Usage minimum (button 1)
29 10 	Usage maximum (button 16)
95 10 	Report count (16) // 16 buttons
75 01 	Report size (1)   // size of field is 1 bit
81 02 	INPUT(Data, Var, Abs) -----------------------------------
06 00 ff Usage page (Vendor defined page)
a1 02 	Collection (logical)
75 08 		Report size (8)
95 07 		Report count (7)
46 ff 00 	Physical Maximum(65280)
26 ff 00 	Logical Maximum(65280)
09 01 		Vendor usage 1
91 02 		Output(Data, Var, Abs)
c0 		End collection 
c0    End collection 
05 0c Usage page Consumer Devices
09 01 Usage device Consumer control
a1 01 Collection (application)
85 03 ???????????????????
05 01 Usage page Generic Desktop
09 06 ???????????????????
a1 02 Collection (logical)
05 06 Usage page ???????????????????
09 20 Usage ???????????????????
15 00 Logical Minimum (0)
26 ff 00 	Logical Maximum(65280)
75 08 Report size (8)
95 01 Report count (1)
81 02 INPUT(Data, Var, Abs)
06 bc ff 0a ad bd 
75 08 Report size (8)
95 06 Report count (6)
81 02 INPUT(Data, Var, Abs)
c0 		End collection 
c0    End collection 
05 0c Usage page Consumer Devices
09 01 Usage device Consumer control
a1 01 Collection (application)
85 ff ???????????????????
05 06 Usage page ???????????????????
95 01 Report count (1)
75 02 Report size (2)
19 24 FLIGHT YOKE
29 26 Usage maximum (button 38)
81 02 INPUT(Data, Var, Abs)
75 06 Report size (6)
81 01 INPUT(Const, Arry, Abs)
c0    End collection

marekhalmo, 2016-09-07:
The important part is here:
05 09 Usage page Button ---------------------------------------
19 01 Usage minimum (button 1)
29 10 Usage maximum (button 16)
95 10 Report count (16) // 16 buttons
75 01 Report size (1) // size of field is 1 bit
81 02 INPUT(Data, Var, Abs) -----------------------------------

which should be approx the same what is in the Joystic.cpp.. so this is a dead end.

I'm trying to read HID descriptor for my Bluetooth gamepad (for android) but i have trouble to install it on my PC.

If you have an android gamepad that can be connected to a PC i would like to see the HID descriptor so i can try to fix this..

marekhalmo, 2016-09-08:
WOW, i finally got it to work! I'm so happy :D

The trick is to set the line 32
0x09, 0x04, // USAGE (Joystick)
to 
0x09, 0x05, // USAGE (Gamepad)

Everything works as before on windows.. the only change is that Android detects the button 0 as Buttoon A and so on...

Could you make the change in the code?

MHeironimus, 2016-09-08:
I am glad to hear you got it to work the way you wanted. Based on your research, I am thinking I will add an option to Version 2.0 that lets you specify the type of Joystick (i.e. Joystick [0x09, 0x04], Gamepad [0x09, 0x04], or Multi-axis Controller [0x09, 0x08]). I will this this issue open until I have had time to add this new feature.

Thanks again for figuring this out,
Matt

marekhalmo, 2016-09-08:
Glad i could help. Thank you for your awesome code that made my project possible!

MHeironimus, 2016-09-14:
@marekhalmo I just uploaded Version 2.0.1 (still considering this in BETA) of the Arduino Joystick Library that adds a constructor parameter that allows you to specify the "type" of joystick you are creating: Joystick (DEFAULT - `JOYSTICK_TYPE_JOYSTICK` or 0x04), Gamepad (`JOYSTICK_TYPE_GAMEPAD` or 0x05), or Multi-axis Controller (`JOYSTICK_TYPE_MULTI_AXIS` or 0x08). If you get a chance, try it out and let me know if it worked for what you were wanting to use it for.

marekhalmo, 2016-09-14:
Thank you i will check this later. Is the v2 stable?

MHeironimus, 2016-09-14:
Version 2.x is stable at this point. I just want to do a little more testing and documentation before I consider it non-beta.

closed 2016-11-16 by MHeironimus