MHeironimus/ArduinoJoystickLibrary #19: Multiple Single Joysticks change enumeration order on windows

zaped212, 2016-09-22:
Not sure if "Issues" is the correct place for this but i wasnt finding a better place.

I have been working on a project that requires 4 "independent" joysticks in the system. Eventually I plan on moving to your Joystick3 class and extend it to 4 controllers though for now I would like to see if there is a way that i could get the 4 different joysticks ( Each different Leonardo + USB connection ) to enumerate in the same order each time.

I noticed in the Joystick2/3 there was a JOYSTICK2_REPORT_ID. Does that help control order of enumeration / allow windows to see each joystick as a different device and hopefully maintain enumeration or is there a better way you know of.

Worst case i think the joystick3 class will work ( once extended to 4 and adding some of the buttons back in ) but thought i would ask your opinion.

MHeironimus, 2016-09-22:
@zaped212 Take a look at Version 2.0 of the library. It supports 4 (or more) joysticks and allows you to specify the REPORT_IDs that are used. You will find Version 2.0 at https://github.com/MHeironimus/ArduinoJoystickLibrary/tree/version-2.0. 

Let me know if that version addresses your issues.

zaped212, 2016-09-26:
I checked out version 2 of your library and was a nice direct drop in replacement for the older version ( very nice ). I know i would like to go with your version 2 from here on, though the REPORT_IDs still confuse me.

In my setup I have 4 Arduino Leonardo's hooked up to a PC and they must maintain their Joystick enumeration number across power cycles. Currently All 4 joysticks have the same report ID ( version 1 ). Version 2 will allow me to easily give each joystick its own report ID but will that be enough for Windows to continually re-enumerate the joystick to the same ID ( Joystick1, 2, 3, 4 ) or are the report ID's only useful through a single power cycle of the PC?

MHeironimus, 2016-09-26:
I have implemented a 3 controller joystick with the library (see http://mheironimus.blogspot.com/2015/12/ultimate-classic-game-console-joystick.html for more details) and the individual controllers have always enumerated the same way (in REPORT_ID order). I do not know that I would go so far as to say I would guarantee that is the case (I have not done the research I need to do to say that), but it has been my experience that they will stay in the same order.

zaped212, 2016-09-27:
With a single USB connection I would expect it to enumerate in the same order since that is the order it will always receive the descriptor table.

I think I should probably move to the single USB connection ( sooner rather than later ) otherwise i look to be susceptible to power up sequencing on different boot cycles.

Thanks for getting back to me on this, Loving the Library!!!

closed 2016-09-28 by zaped212