MHeironimus/ArduinoJoystickLibrary #24: Arduino Micro Issues

loriana, 2016-12-05:
Have tried to use the example sketches with arduino micro and they compile and upload, but the arduino is not recognised as a joystick by the computer for some reason. Used the older library before and that worked out well, this one would suit my needs better though.

I'm using Arduino 1.6.12 and Windows 7.

MHeironimus, 2016-12-06:
@loriana I do not currently have an Arduino Micro available to test with, but I will try to get one in a couple of weeks and see if I can reproduce your issue. I will let you know what I find.

MHeironimus, 2016-12-20:
@loriana - My Arduino Micro should arrive in a few days. As soon as it arrives I will see if I can reproduce your issue. I will let you know what I find.

MHeironimus, 2016-12-20:
@loriana I just tested the library with a new Arduino Micro using Arduino 1.6.13 and Windows 10 and it worked OK. I am not sure why it is not working for you. Are you sure the new sketch is successfully loading on your board? Can you post the compile / upload log?

MHeironimus, 2016-12-26:
@loriana  Since I could not reproduce this issue, I am going to close this out. If you are able to reproduce the problem and upload the logs, I can reopen it.

closed 2016-12-26 by MHeironimus