MHeironimus/ArduinoJoystickLibrary #25: Reversed axis range could not be defined

baritonomarchetto, 2016-12-15:
First thing first: this is not really an issue but more a request.

I was toying with the new library version and tried to reverse the axis behaviour. My starting code was this:

[code]
#include <Joystick.h>

//pure x axis...
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_JOYSTICK,
  0, 0,                  
  true, false, false,    
  false, false, false,   
  false, false,          
  false, false, false);  

int readPot;

void setup() {
 Joystick.begin();
 Joystick.setXAxisRange(0, 1023);
}

void loop() {
readPot = analogRead(A2);

 Joystick.setXAxis(readPot);
}
[/code]

This code works flawlessy, but if you set a reversed XAxisRange:

[code]
 Joystick.setXAxisRange(1023, 0);
[/code]

It "doens't work": the axis cannot be controlled by the pot anymore. 
I know that this could be "fixed" by a simple, standard, arduino "map" function, but it would be nice to reverse the axis by simply reverse the range definition :)

Keep the great work up!

MHeironimus, 2016-12-20:
@baritonomarchetto - I created a new branch (issue-25) that I think implements the enhancement you have requested. I have not fully tested it yet, but it did pass my basic JoystickTest sketch. If you get a chance can you download this version (https://github.com/MHeironimus/ArduinoJoystickLibrary/tree/issue-25) and verify it does what you wanted?

Thanks,
Matt

baritonomarchetto, 2016-12-21:
Thanks! I will try as soon as possible and report back

baritonomarchetto, 2016-12-21:
Matthew, I made my homework and I can say that if i define an inverted range like 

[code] Joystick.setXAxisRange(1023, 0); [/code]

it works as it should!!

There's some issue unfortunately:

1) If you define the "straight" reduced range (823 200) the axis limits are reached with a reduced rotation of the pot, like it was in the V2.0.1; if you define a reduced inverted range instead... 

[code] Joystick.setXAxisRange(823, 200); [/code]

the cross reaches the axis limit and goes throught the other limits.

2) If you define a range like 

[code] Joystick.setXAxisRange(-127, 127); [/code]

(which is the V1 standard IIRC) only the right semiaxis could be sent... 

(Thank you very much!)

MHeironimus, 2016-12-21:
The way I designed it, values specified by a call to `setXAxis` that are outside the range specified by `setXAxisRange` should be ignored. For example, if you set the range using `Joystick.setXAxisRange(823, 200)`, but then did a `setXAxis(125)`, the call to `setXAxis` should be ignored. Based on your description, it sounds like this might not be working correctly. I will try to take a look at the scenarios you describe above and see if I can fix the issue.

I probably will not get to this for a few days, however. I will let you know.

Thanks again for testing this,
Matt

baritonomarchetto, 2016-12-21:
I am not in a hurry, take your time and thanks for the support!

MHeironimus, 2016-12-24:
@baritonomarchetto Thanks again for helping out with the testing of this change. I think I have those two issues you identified resolved. Could you download the latest version from the issue-25 branch (https://github.com/MHeironimus/ArduinoJoystickLibrary/tree/issue-25) and let me know if I got it right this time?

I also changed how input values outside the expected range are handled. If a value smaller than the minimum value is encountered, the minimum value is used (instead of just ignoring the input). If a value larger than the maximum value is encountered, the maximum value is used (instead of just ignoring the input).

Thanks again for your help,
Matt

baritonomarchetto, 2016-12-25:
Matt (or should I call you... Santa? :) ) works flawlessy now!

Thanks for your efforts and help!

baritonomarchetto, 2016-12-26:
Made some additional test, and unfortunately the negative definition is still in the need for some tweaking IF AND ONLY IF defining a negative range makes still sense.
To be more precise, by defining the axis range in the sketch below:

[code]
#include <Joystick.h>

//pure x axis...
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_JOYSTICK,
0, 0,
true, false, false,
false, false, false,
false, false,
false, false, false);

int readPot;

void setup() {
Joystick.begin();
Joystick.setXAxisRange(0,1023); //full range - WORKS
//Joystick.setXAxisRange(1023,0); //inverted full range - WORKS
//Joystick.setXAxisRange(400,623); // reduced straigh ranhe - WORKS
//Joystick.setXAxisRange(0,1023); //inverted reduced range - WORKS
//Joystick.setXAxisRange(-127,127); //reduced negative range - DOESNT WORK
}

void loop() {
readPot = analogRead(A0);
Joystick.setXAxis(readPot);
}
[/code]

... everithing works as it should.

By defining a negative range...
[code]
Joystick.setXAxisRange(-127,127); //full range
[code] 
only half axis is available.

MHeironimus, 2016-12-26:
@baritonomarchetto Per the Arduino documentation the `analogRead` will only return a value from `0` to `1023` (see https://www.arduino.cc/en/Reference/AnalogRead). Based on that, your example above should only go from the center to the right (you would never be able to get to the left of center).

I created a simple test sketch (see below) that demonstrates a value swinging from -130 to 130 with the range set to -127 to 127. This example works the way I would expect. Let me know if you still have questions about negative ranges. 

~~~
#include <DynamicHID.h>
#include <Joystick.h>

/*
  Ground pin A0 to activate the test.
*/

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_JOYSTICK, 4, 0,
  false, false, false, false, false, false,
  false, false, false, false, true);

// the setup routine runs once when you press reset:
void setup() {
  
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB
  }

  Joystick.setSteeringRange(-127, 127);
  Joystick.begin();

  // Init Pins
  pinMode(A0, INPUT_PULLUP);
}

int counter = -130;
bool goingUp = true;

// the loop routine runs over and over again forever:
void loop() {
  
  // read the input on analog pin 0:
  if (!digitalRead(A0)) {

    Serial.println(counter);
    Joystick.setSteering(counter);
    
    if (goingUp) {
      counter++;
      if (counter > 130) {
        counter = 129;
        goingUp = false;
      }
    } else {
      counter--;
      if (counter < -130) {
        counter = -129;
        goingUp = true;
      }
    }
  }

  delay(5);
}
~~~

baritonomarchetto, 2016-12-26:
Ah,  ok,  make sense to me now. Thank you!

MHeironimus, 2016-12-26:
This change has now been merged into the master and version-2.0 branches.

closed 2016-12-26 by MHeironimus