MHeironimus/ArduinoJoystickLibrary #34: i got wrong value in unity

rswa000, 2017-01-23:
hello ..  MHerionimus..
i love your adrduinoJoystickLibraray

i test axis in the version 2.2
i got correct value in DIView, but it's wrong in unity 5.5..
i need to map 0 ~ 1023  to  512 ~ 1023, then i got correct value in unity..
i cant figure out what happen....
could you help me?

MHeironimus, 2017-02-08:
In order to assist you I will need some code samples I can run that duplicate your issue. For example, I will need the sketch file you are using on the Arduino Leonardo. I will also need the source code for your DIView example that is working and the Unity example that is not working.

SoylentGraham, 2017-03-14:
@rswa000 I'm having a problem in unity, on windows too. My axis always reports -1 instead of 0 when idle. (Then 1 when it's right, which is correct).
Not a problem on OSX though.

Can you explain your fix? what did you change from 0-1023 to 512-1023? (min? max? the defaults are none of these values as far as I can see?)

rswa, 2017-03-15:
map is a function in Arduino.
if ur range is 0 - 1023, u need to map input to 512 - 1023.
u just need the upper half range.
I didn't fix it.
I just follow the output rule.

MHeironimus, 2017-09-03:
This issue may not be unique to the Arduino Joystick Libarary: [https://stackoverflow.com/questions/43591045/unity-compatible-joystick-for-windows-10](https://stackoverflow.com/questions/43591045/unity-compatible-joystick-for-windows-10)

wormyrocks, 2017-10-17:
I'm also having this issue - trying to send quaternion values from an IMU over HID joystick axes with an ATMega32U4. Identical Unity code works perfectly on OSX and fails under Windows 10.

MHeironimus, 2017-10-28:
I have not used Unity, but I am hearing a lot of people report issues using joysticks created with this library and also trying to use generic joysticks and gamepads with Unity. I am not sure, but my current theory is Unity has an issue with raw input values that are negative. 

The way the Joystick library works, it always sends -32767 for the minimum axis value and 32767 for the maximum. I think this issue could be corrected by modifying the library to use 0 for minimum and 32767 (or maybe 65535) for the maximum.

It may take me awhile, but if I can find some time, I can try to create a new version that goes from 0 to 65535 and we can see if that addresses the issue.

Since this only affects one development platform (and I think the issue is actually in Unity), this is not my highest priority, but I will try to get to it.

JuanJSAR, 2018-09-21:
> 
> 
> ...The way the Joystick library works, it always sends -32767 for the minimum axis value and 32767 for the maximum. I think this issue could be corrected by modifying the library to use 0 for minimum and 32767 (or maybe 65535) for the maximum...
Could you indicate how to do it?

> 
> 
>...It may take me awhile, but if I can find some time, I can try to create a new version that goes from 0 to 65535 and we can see if that addresses the issue...
It would be excellent if I could do it because I have the same problem.

MHeironimus, 2018-09-21:
There are a few things that would need to be changed:

- The HID report descriptor section that indicates the range for analog axis

~~~
...
	if (axisCount > 0) {

		// USAGE (Pointer)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;

		// LOGICAL_MINIMUM (-32767)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x16;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x80;

		// LOGICAL_MAXIMUM (+32767)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x26;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0xFF;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x7F;

		// REPORT_SIZE (16)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x75;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x10;
...
~~~

- The `buildAndSet16BitValue` function would need to be updated.

nx-sm, 2018-10-25:
> Could you indicate how to do it?
>
>
> It would be excellent if I could do it because I have the same problem.

Hi, I am having the same problem. Have you managed to fix it by any chance? I don't understand what values need to be changed. Cheers.

gregnau, 2019-03-08:
@JuanJSAR && @nx-sm 

I have made a proper 16 bit unsigned version of the Joystick Library as @MHeironimus suggested in his post. 100 % same lib, but sending values between 0-65535.
[http://github.com/gregnau/ArduinoJoystickLibrary](http://github.com/gregnau/ArduinoJoystickLibrary)

gregnau, 2019-03-31:
After a few weeks of experimenting with this unsigned mod, it seems like maybe not the best solution. Worked like a charm under Windows 10 (1803) and OSX, but had serious issues on Ubuntu (16.04).
It is expecting values between -32767 and 32767, while the proper range is transmitter with the usb descriptor.
Modifying the range used in the library to 0-32767 and setting the descriptors according solved the issue and now it works for sure on Windows and Linux. Both Unity and a lots of other games and sims.

mozzis, 2020-04-16:
Seems like [http://github.com/gregnau/ArduinoJoystickLibrary](http://github.com/gregnau/ArduinoJoystickLibrary) no longer exists, what happened?

martinling, 2020-10-27:
A working version of the fix for this is in this commit from @yiktungho: https://github.com/yiktungho/ArduinoJoystickLibrary/commit/f3585953a6f262083e5d3cb0ad0710829723afdc

It appears that this is not just a Unity problem, but affects any program accessing the joystick through the RawInput API on Windows. I was able to reproduce it on Star Wars Battlefront II (Frostbite engine), for instance.

Using [this test program](http://www.planetpointy.co.uk/joystick-test-application/), you can select to access the device through either RawInput or DirectX. 

With the -32768 to 32767 range, correct movements are shown with the DirectX method but not with RawInput.

With the change to a 0 to 32767 range, correct movements are shown for both methods.

James2k2, 2021-06-10:
So the modification from gregnau isn't around anymore, so I've done a modification myself that uses a range of 0 - 65535: https://github.com/James2k2/ArduinoJoystickLibrary

Hope this helps anybody else out! We've tested this with Unity and it works. I've renamed it as well so you should be able to install it to the Arduino IDE like any other and it's called Joystick2. A extra example is included to demonstrate the new range.

MHeironimus, 2022-03-30:
Version 2.1.0 of the library should address the issues seen in Unity (and other applications).

closed 2022-03-30 by MHeironimus