MHeironimus/ArduinoJoystickLibrary #36: Any ARM board supported?

leonardosalvatore, 2017-02-08:
Hi,
looking to use this fantastic library for a more CPU demanding project project I'm asking if any ArduinoMKRZero or Teensy  board is supported or tested.
Sorry for rise this as an issue.
Thanks.

MHeironimus, 2017-02-08:
I have not tested with either of the boards listed. It should work with any board that uses a ATmega32u4 microcontroller. 

The Teensy 2.0 (https://www.pjrc.com/teensy/) uses a ATmega32u4, so it should theoretically work. 

The ArduinoMKRZero uses a SAMD21 Cortex-M0+ microcontroller (https://www.arduino.cc/en/Main/ArduinoBoardMKRZero). I have not tested any boards that use this microcontroller. I did just try to compile one of the test sketch files targeting the ArduinoMKRZero and it did not work.

leonardosalvatore, 2017-02-09:
Thank very much for this valuable information.

closed 2017-02-09 by leonardosalvatore