MHeironimus/ArduinoJoystickLibrary #37: Arduino Mega

sreksten, 2017-02-10:
Well, this is acually not a real Issue, so maybe mr. Heironimus would prefer to move this on the documentation, or whatever.

I would like to report my findings to everyone interested on using the ArduinoJoystickLibrary on the Arduino Mega. Why the Mega? Well, with so many pins, you can build a double controller binding leds to certain buttons, and that is what I wanted to build an arcade cabinet using as less hardware as possible.

It is definitely possible to use it!

1) You have to flash the NicoHood HoodLoader2 on the 16u2 of the Mega. It is a very simple process if you follow the instructions.
2) Due to the fact that the 16u2 has *very* limited resources, you must stick to version 1.0 of the Joystick library.  I tried to use version 2.0, but the Arduino IDE warns you that resources will be running low, and the 16u2 will simply brick :) So I had to flash HoodLoader2 again and restart from scratch. Maybe a crimped version of the 2.0 would work but I didn't test this.
3) You have to write a small "communication bridge" program between the 16u2 and the AtMega328. It will simply get bytes (in a protocol you have to decide) from the AtMega328, and transform them in calls to the Joystick library.
3) Write an arbitrary complex program on the AtMega328 that will allow you to have some complex behaviour, like button rebinding, autofire, and so on. The program will have to transform data so that the 16u2 will understand them.
4) If you use Windows, it will see 2 controllers and you're done! If you use Linux, it will see the double controller as a single one, and you have to pass the correct parameter to the kernel, as documented by mr. Heironimus (the HID_QUIRK_MULTI_INPUT issue). But, as you have flashed the 16u2 with HoodLoader2, you have to change the Product ID with 0x484C. That is, add "usbhid.quirks=0x2341:0x484C:0x040" to /boot/cmdline.txt (I determined the value plugging the Mega under Window and checking it in the Control Panel).

I managed to plug such a modified version of the Mega to a Raspberry Pi 3 and Retropie sees it as a double controller, that actually works! :)

So, I'd like to thank mr. Heironimus and mr. Nico Hood for their work, and I hope that this post may help someone else.

MHeironimus, 2017-02-17:
Thank you for the great comments and input. When I get some time, I will incorporate this into the documentation for Version 1.0.

WillDrug, 2020-12-31:
You are a life saver, I've missed that titbit and now I sit here with a fully built system which can't do what I want it to D: 
Thank you for that one