MHeironimus/ArduinoJoystickLibrary #38: Arduino Pro Micro

JPG-Consulting, 2017-02-17:
I'm having some problems here...

    ccK89KZw.ltrans0.ltrans.o: In function `__base_ctor ':
    Joystick.cpp:439: undefined reference to `DynamicHID()'
    undefined reference to `DynamicHID_::AppendDescriptor(DynamicHIDSubDescriptor*)'
    collect2.exe: error: ld returned 1 exit status
    exit status 1

Editing "joystick.h" and changing 

    #include <DynamicHID/DynamicHID.h>

with 

    #include "DynamicHID/DynamicHID.h"

Solves that issue but brings new issues:

    cKCayaX.ltrans0.ltrans.o: In function `__base_ctor ':
    libraries\joystick/Joystick.cpp:439: undefined reference to `DynamicHID()'
    libraries\joystick/Joystick.cpp:439: undefined reference to `DynamicHID_::AppendDescriptor(DynamicHIDSubDescriptor*)'
    collect2.exe: error: ld returned 1 exit status
    exit status 1

MHeironimus, 2017-02-17:
Can you post reply with the following information?
- What board are you targeting?
- What version of the Arduino IDE are you using.
- What operating system are you running the Arduino IDE on?
- What version of the Joystick library are you including?

Thanks,
Matt

JPG-Consulting, 2017-02-18:
Board: Pro Micro
IDE Version: 1.8.1
OS: Windows 10
Joystick version: master

I finally it "working"... but only top and lower values are received. For example, setting throttle range of 0-255 will only reflect 0 or 255... If i do `setThrottle(132);` my OS refects throttle it at it's maximum when it is slightlly over half full throttle. Same thing happens to all other axis

MHeironimus, 2017-03-16:
Can you post your sketch file so I (or someone else in this community) can take a look?

reppners, 2017-09-10:
I've got the same issue here. My project setup is not using the ArduinoIDE directly but rather my editor of choice with PlatformIO for the toolchain / setup.

* Arduino Pro Micro board (platformio board `sparkfun_promicro16`)
* macOS
* code from `master`-branch
* PlatformIO, version 3.5.0a9
```
Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.7.1          [Up-to-date]
Updating toolchain-atmelavr              @ 1.40902.0      [Up-to-date]
Updating framework-arduinoavr            @ 1.10619.0      [Up-to-date]
Updating tool-avrdude                    @ 1.60300.2      [Up-to-date]
```

Somehow the DynamicHID part was not compiled / included in compilation.

```c++
#include <Arduino.h>
#include <Joystick.h>

Joystick_ Joystick = Joystick_(
  0x03,
  JOYSTICK_TYPE_GAMEPAD, // device type
  10,                     // button count
  0,                     // hat-switch count
  true,                  // includeXAxis
  true,                  // includeYAxis
  false,                 // includeZAxis
  false,                 // includeRxAxis
  false,                 // includeRyAxis
  false,                 // includeRzAxis
  false,                 // includeRudder
  false,                 // includeThrottle
  false,                 // includeAccelerator
  false,                 // includeBrake
  false);                // includeSteering
```

led to

```bash
[Sun Sep 10 12:53:24 2017] Processing sparkfun_promicro16 (platform: atmelavr; board: sparkfun_promicro16; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Collected 27 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Joystick> (/Users/blablub/Development/Arduino/N64_Controller/Test/lib/Joystick)
avr-g++ -o .pioenvs/sparkfun_promicro16/firmware.elf -Os -mmcu=atmega32u4 -Wl,--gc-sections -flto -fuse-linker-plugin .pioenvs/sparkfun_promicro16/src/main.o .pioenvs/sparkfun_promicro16/src/n64-controller/n64_controller.o -L.pioenvs/sparkfun_promicro16 -Wl,--start-group .pioenvs/sparkfun_promicro16/libFrameworkArduinoVariant.a .pioenvs/sparkfun_promicro16/libFrameworkArduino.a -lm .pioenvs/sparkfun_promicro16/lib/libJoystick.a -Wl,--end-group
/var/folders/wh/30qp97b17hjb8_tp46mj7wzh0000gn/T//ccjN3LPZ.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_main.o.2195':
ccjN3LPZ.ltrans0.o:(.text.startup+0x20c): undefined reference to `DynamicHID()'
ccjN3LPZ.ltrans0.o:(.text.startup+0x212): undefined reference to `DynamicHID_::AppendDescriptor(DynamicHIDSubDescriptor*)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sparkfun_promicro16/firmware.elf] Error 1
```

To solve the issue I copied the DynamicHID folder to the `lib` folder and added an explicit include statement.

```c++
#include <Arduino.h>
#include <DynamicHID.h>
#include <Joystick.h>

Joystick_ Joystick = Joystick_(
  0x03,
  JOYSTICK_TYPE_GAMEPAD, // device type
  10,                     // button count
  0,                     // hat-switch count
  true,                  // includeXAxis
  true,                  // includeYAxis
  false,                 // includeZAxis
  false,                 // includeRxAxis
  false,                 // includeRyAxis
  false,                 // includeRzAxis
  false,                 // includeRudder
  false,                 // includeThrottle
  false,                 // includeAccelerator
  false,                 // includeBrake
  false);                // includeSteering
```

```bash
[Sun Sep 10 12:55:51 2017] Processing sparkfun_promicro16 (platform: atmelavr; board: sparkfun_promicro16; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Collected 27 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Joystick> (/Users/blablub/Development/Arduino/N64_Controller/Test/lib/Joystick)
|-- <DynamicHID> (/Users/blablub/Development/Arduino/N64_Controller/Test/lib/DynamicHID)
avr-g++ -o .pioenvs/sparkfun_promicro16/src/main.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DPLATFORMIO=30500 -DARDUINO_ARCH_AVR -DARDUINO_AVR_PROMICRO16 -DARDUINO=10619 -DUSB_VID=0x1B4F -DUSB_PID=0x9203 "-DUSB_PRODUCT=\"SparkFun Pro Micro\"" -DUSB_MANUFACTURER=\"SparkFun\" -I/Users/blablub/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/blablub/.platformio/packages/framework-arduinoavr/variants/sparkfun_promicro -Isrc -Ilib/Joystick -Ilib/DynamicHID src/main.cpp
avr-g++ -o .pioenvs/sparkfun_promicro16/firmware.elf -Os -mmcu=atmega32u4 -Wl,--gc-sections -flto -fuse-linker-plugin .pioenvs/sparkfun_promicro16/src/main.o .pioenvs/sparkfun_promicro16/src/n64-controller/n64_controller.o -L.pioenvs/sparkfun_promicro16 -Wl,--start-group .pioenvs/sparkfun_promicro16/libFrameworkArduinoVariant.a .pioenvs/sparkfun_promicro16/libFrameworkArduino.a -lm .pioenvs/sparkfun_promicro16/lib/libJoystick.a .pioenvs/sparkfun_promicro16/lib/libDynamicHID.a -Wl,--end-group
avr-size --mcu=atmega32u4 -C -d .pioenvs/sparkfun_promicro16/firmware.elf
avr-objcopy -O ihex -R .eeprom .pioenvs/sparkfun_promicro16/firmware.elf .pioenvs/sparkfun_promicro16/firmware.hex
AVR Memory Usage
----------------
Device: atmega32u4

Program:    5686 bytes (17.4% Full)
(.text + .data + .bootloader)

Data:        280 bytes (10.9% Full)
(.data + .bss + .noinit)

=============================================================== [SUCCESS] Took 1.16 seconds ==============================================================
```

closed 2018-02-24 by MHeironimus