MHeironimus/ArduinoJoystickLibrary #39: OSX only sees one joystick. But all Axis' & buttons

SoylentGraham, 2017-03-13:
I'm using a Pro Mini clone (still using a ATMega34U - https://www.amazon.co.uk/gp/product/B019SXN84E/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1 ), in case that has any bearing;

I have implemented multiple joysticks on one arduino, with multiple buttons, multiple axis';
On Windows these show up as separate joysticks (all working)
On OSX (Sierra) the HID system sees 1 joystick, with all the axis' and buttons present (and working)
Tested with unity and an app from the MacApp store; "Controllers Lite"

Any idea where this could be going wrong? Maybe OSX has some quirks with the USB-HID protocol?

Untested on linux (although plugging into an android phone showed 1 joystick, but I can't vouch for the test apps I was using, and they may only support one joystick anyway)

MHeironimus, 2017-03-16:
I do not have a Mac to test this out on, but I know in Linux you have to "tweek" the usbhid driver to support single USB devices that appear as multiple joysticks. I wrote an article about this at http://mheironimus.blogspot.com/2015/09/linux-support-for-arduino-leonardo.html. I wonder if you have to do something similar in OSX.

SoylentGraham, 2017-03-28:
I spent a while digging into this... but couldn't really find an equivalent. 
Hacky work around in unity for now until I delve back into HIDManager development

MHeironimus, 2017-03-30:
I did a little searching out on the internet on this issue. Seems like my library may not be the only multiple HID device out there that has issues with Mac OSX:
[http://www.microchip.com/forums/m706502.aspx](http://www.microchip.com/forums/m706502.aspx)
[https://lists.apple.com/archives/usb/2014/Jan/msg00005.html](https://lists.apple.com/archives/usb/2014/Jan/msg00005.html)
[https://github.com/NicoHood/HID/issues/14](https://github.com/NicoHood/HID/issues/14)

There is not much out there on this topic, however. If I every get my hands on a Mac, I can play around with it.

twardnw, 2017-12-05:
FWIW, I've had good results using https://github.com/NicoHood/HID/ , though this library was much easier to implement.

BleuLlama, 2018-05-19:
I'm seeing this issue as well, I've found that i can try to get two joysticks if i have one as JOYSTICK_TYPE_GAMEPAD, and one as JOYSTICK_TYPE_JOYSTICK, but both have the same content. 

@MHeironimus - I have an older macbook i'd be willing to lend to you if you're interested. The battery no longer works, but it runs the arduino IDE just fine, etc.

alijani1, 2021-02-24:
I had to also switch to https://github.com/NicoHood/HID/ to get Mac working well with multiple gamepads :(

with this library, while you can get it to partially work with one joystick set to JOYSTICK_TYPE_GAMEPAD and another as JOYSTICK_TYPE_JOYSTICK, you quickly realize that data overlaps and information is being sent from different game pads on top of each other and OS does not properly understand all the data correctly.

https://github.com/NicoHood/HID/ provides better results but it is not as easy to use as this library.  Hope you can fix this @MHeironimus