MHeironimus/ArduinoJoystickLibrary #40: ButtonState Preference

b3ck, 2017-03-27:
I have a 3 button game pad here is my code:

`#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD,
  3, 0,                  // Button Count, Hat Switch Count
  false, false, false,   // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering

void setup() {
  // Initialize Button Pins
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin();
}

// Constant that maps the phyical pin to the joystick button.
const int pinToButtonMap = 2;

// Last state of the button
int lastButtonState[3] = {0,0,0};

void loop() {

  // Read pin values
  for (int index = 0; index < 3; index++)
  {
    int currentButtonState = !digitalRead(index + pinToButtonMap);
    if (currentButtonState != lastButtonState[index])
    {
      if (index < 4) {
        Joystick.setButton(index, currentButtonState);
        lastButtonState[index] = currentButtonState;
      } else {
        if (currentButtonState) {
          Joystick.setButton(index, currentButtonState);
          lastButtonState[index] = currentButtonState;
        }
      }
    }
  }

  delay(10);
}`

My button on pin 2 is normally on, how do I reverse this in the code so that the state is normally off.

Thanks in advance,
-b3ck

SoylentGraham, 2017-03-28:
If this is what you currently have...
`int currentButtonState = !digitalRead(index + pinToButtonMap);`
Isn't that just the part you need to reverse?

Perhaps slightly more clear code would help
`int currentButtonState = digitalRead(index + pinToButtonMap) == HIGH;`

b3ck, 2017-03-28:
I really have no clue, I just modified the the keyboard + button joystick code to get it to work properly, I basically just need to be able to control each of the buttons initial state, because sometimes the switches/buttons might be reversed.

At first I thought it could be done with this part:
```
// Last state of the button
int lastButtonState[3] = {0,0,0};
```
Where I would just change this "{0,0,0}" to like this "{0,2,0}" then my button on pin 3 would be normally on or in "HIGH" as you put it. But nope.  Basically I just need to be able to control each of the 3 button states on the fly because I never know how a switch/button is going to react from my huge bucket of switches/buttons.

Thanks in advance,
-b3ck

b3ck, 2017-03-30:
Anyone have any ideas on how I would do this?

-b3ck

SoylentGraham, 2017-03-30:
Your question is a bit too vague I think, and it's not really clear what you're trying to achieve.
`lastButtonState` IS your initial state. But every update, you basically change it, according to your physical button/pin state.
Your use of `lastButtonState` just checks for a change from the previous loop, the values you initialise `lastButtonState` with are kind of irrelevant.

When the pin for each button is high (physical button is being pressed)... what do you want your corresponding joystick button to do?

If you just want the joystick to say what the physical button is doing, just do
```for (int index = 0; index < 3; index++)
  {
    int currentButtonState = digitalRead(index + pinToButtonMap) == HIGH;
    Joystick.setButton(index, currentButtonState);
}
```
You don't need `lastButtonState` at all to make your "joystick" reflect what your buttons are doing.
This question is better suited to [stackoverflow](http://www.stackoverflow.com) as it's not really related to the code in this repository.

b3ck, 2017-03-30:
Okay I am sorry I will delete it, thanks.

closed 2017-03-30 by b3ck