MHeironimus/ArduinoJoystickLibrary #41: Changing pins in GamepadExample triggers up randomly

jbenam, 2017-04-05:
Hi, thanks for your work on this library - much appreciated!

I've recently bought an Arduino Leonardo just for this project (genuine one, not a clone, in case it matters) but I'm having some trouble changing the pins from GamepadExample.

If for example I just change pins 2 to 6 with pins 3 to 7, the emulated pad starts going bananas with up being randomly triggered (touching the Arduino makes it trigger more often). And that's with NOTHING connected to it! I'm kinda new to Arduino, so I might be missing something terribly straightforward.

The issue is only exacerbated by adding more buttons. Then it starts doing all kinds of funny stuffs (triggering button 0 repeatedly, going down-left like mad, etc). The only thing that seems to be working right is a vanilla GamepadExample - as soon as I change the pins, it goes totally nuts. Maybe my Arduino is faulty?

Thanks for your help!

MHeironimus, 2017-04-06:
Please post your sketch file and describe how you are wiring up the buttons.

jbenam, 2017-04-06:
Hi Matthew, thanks for your reply.

I'm just using the GamepadExample sketch you've provided.

The only thing I've changed are the 5 pinModes:

`  // Initialize Button Pins
  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
  pinMode(6, INPUT_PULLUP);
  pinMode(7, INPUT_PULLUP);`

The issue happens without anything wired to it, that's the worst part! Doesn't happen if I just upload the untouched example, though.

Buttons (and the stick) are just wired up with a common ground to GND and the other wire to the various digital pins.

Thanks again.

SoylentGraham, 2017-04-06:
When nothing is connected to an arduino pin, you cannot guarantee what the voltage/digital reading of that pin will be.

jbenam, 2017-04-06:
Right, I know as much - but I've tested it with everything connected to it and the issue still manifests itself in the same exact way.

BleuLlama, 2017-04-06:
He used the INPUT_PULLUP on them, which enables the weak pull-up in the Avr, which should be sufficient/equivalent to an external weak pull-up resistor

MHeironimus, 2017-04-06:
@jbenam Can you post your entire sketch file? The issue may be in how you are processing the inputs or how you are sending them to the Joystick library.

jbenam, 2017-04-06:
@MHeironimus 
Hi Matthew, it's just your example code with a few things changed. Here it is:
`#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD,
  1, 0,                  // Button Count, Hat Switch Count
  true, true, false,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering

void setup() {
  // Initialize Button Pins
  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
  pinMode(6, INPUT_PULLUP);
  pinMode(7, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin();
  Joystick.setXAxisRange(-1, 1);
  Joystick.setYAxisRange(-1, 1);
}

// Last state of the buttons
int lastButtonState[5] = {0,0,0,0,0};

void loop() {

  // Read pin values
  for (int index = 0; index < 5; index++)
  {
    int currentButtonState = !digitalRead(index + 2);
    if (currentButtonState != lastButtonState[index])
    {
      switch (index) {
        case 0: // UP
          if (currentButtonState == 1) {
            Joystick.setYAxis(-1);
          } else {
            Joystick.setYAxis(0);
          }
          break;
        case 1: // RIGHT
          if (currentButtonState == 1) {
            Joystick.setXAxis(1);
          } else {
            Joystick.setXAxis(0);
          }
          break;
        case 2: // DOWN
          if (currentButtonState == 1) {
            Joystick.setYAxis(1);
          } else {
            Joystick.setYAxis(0);
          }
          break;
        case 3: // LEFT
          if (currentButtonState == 1) {
            Joystick.setXAxis(-1);
          } else {
            Joystick.setXAxis(0);
          }
          break;
        case 4: // FIRE
          Joystick.setButton(0, currentButtonState);
          break;
      }
      lastButtonState[index] = currentButtonState;
    }
  }

  delay(10);
}`

MHeironimus, 2017-04-09:
@jbenam The only thing I see wrong with your sketch file is the following line:

```
    int currentButtonState = !digitalRead(index + 2);
```

should read 

```
    int currentButtonState = !digitalRead(index + 3);
```

assuming you want pin 3 to be up, pin 4 to be right, 5 to be down, 6 to be left, and 7 to be fire.

I uploaded your sketch (which the change above) to my Leonardo and it worked without error.

You may want to try to get your hands on another Leonardo or Micro and see if you can reproduce the issue.

eadmaster, 2017-04-15:
aren't you supposed to initialize the buttons pins to HIGH as [described here](https://www.sparkfun.com/tutorials/171#setup)?

BleuLlama, 2017-04-15:
They are equivalent. 

Pi

BleuLlama, 2017-04-15:
They are equivalent:
E.g.:

pinMode( pin, INPUT );
digitalWrite( pin, HIGH );

Is the same as:

pinMode( pin, INPUT_PULLUP );

-s

thedrumm, 2017-05-22:
I have to say I got the same issues, but in my case I did some changes for using more buttons.
I uploaded the program into my arduino micro (chinesse) several times; without making any change into the code.
It worked at least really well.

Sylverzerom, 2017-05-22:
this might help with your problem, i used arrays to store the pins.

`// Simple gamepad example that demonstraits how to read five Arduino
// digital pins and map them to the Arduino Joystick library.
//
// The pins are grounded when they are pressed.
// Pin A0 = UP
// Pin A1 = DOWN
// Pin A2 = LEFT
// Pin A3 = RIGHT
// Pin 2 = A
// Pin 3 = B
// Pin 4 = Start
// Pin 5 = Select
// Pin 6 = L Shoulder
// Pin 7 = R Shoulder
// Pin 8 = Modifier

//reserved pins for later
// Pin 9 = X
// Pin 10 = Y
// Pin 11 = L trigger
// Pin 12 = R trigger
//
// by Sylverzerom
// 05-05-2017
//--------------------------------------------------------------------

#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD,
  7, 0,                  // Button Count, Hat Switch Count
  true, true, false,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering

// Last state of the buttons _ number of array is pins used
int lastButtonState[11] = {0,0,0,0,0,0,0,0,0,0,0};
word input_pins[11] = {A0,A1,A2,A3,2,3,4,5,6,7,8};

void setup() {
  // Initialize Button Pins

  for (int index = 0; index < 11; index++)
  {
    pinMode(input_pins[index], INPUT_PULLUP);
  }
//  pinMode(3, INPUT_PULLUP);
  //pinMode(4, INPUT_PULLUP);
  //pinMode(5, INPUT_PULLUP);
  //pinMode(6, INPUT_PULLUP);
  //pinMode(A1, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin();
  Joystick.setXAxisRange(-1, 1);
  Joystick.setYAxisRange(-1, 1);
}

void loop() {
  RXLED0;
  TXLED0;
  
  // Read pin values
  for (int index = 0; index < 11; index++)
  {
    int currentButtonState = !digitalRead(input_pins[index]);
    if (currentButtonState != lastButtonState[index])
    {
      switch (index) {
        case 0: // UP
          if (currentButtonState == 1) {
            Joystick.setYAxis(-1);
          } else {
            Joystick.setYAxis(0);
          }
          break;
        case 1: // DOWN
          if (currentButtonState == 1) {
            Joystick.setYAxis(1);
          } else {
            Joystick.setYAxis(0);
          }
          break;
        case 2: // LEFT
          if (currentButtonState == 1) {
            Joystick.setXAxis(-1);
          } else {
            Joystick.setXAxis(0);
          }
          break;
        case 3: // RIGHT
          if (currentButtonState == 1) {
            Joystick.setXAxis(1);
          } else {
            Joystick.setXAxis(0);
          }
          break;
        case 4: // A
          Joystick.setButton(0, currentButtonState);
          break;
        case 5: // B
          Joystick.setButton(1, currentButtonState);
          break;
        case 6: // Start
          Joystick.setButton(2, currentButtonState);
          break;
        case 7: // Select
          Joystick.setButton(3, currentButtonState);
          break;
        case 8: // L Shoulder
          Joystick.setButton(4, currentButtonState);
          break;
        case 9: // R Shoulder
          Joystick.setButton(5, currentButtonState);
          break;
        case 10: // Modifier
          Joystick.setButton(6, currentButtonState);
          break;
      }
      lastButtonState[index] = currentButtonState;
    }
  }

}

`

jbenam, 2017-05-26:
This is what I ended up doing instead, works pretty well:

```
#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD,
  5, 0,                  // Button Count, Hat Switch Count
  true, true, false,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering

void setup() {
  pinMode(10, INPUT_PULLUP); // UP
  pinMode(11, INPUT_PULLUP); // RIGHT
  pinMode(12, INPUT_PULLUP); // DOWN
  pinMode(13, INPUT_PULLUP); // LEFT
  pinMode(3, INPUT_PULLUP); // 0
  pinMode(4, INPUT_PULLUP); // 1
  pinMode(5, INPUT_PULLUP); // 2
  pinMode(6, INPUT_PULLUP); // 3
  pinMode(7, INPUT_PULLUP); // 4

  // Initialize Joystick Library
  Joystick.begin();
  Joystick.setXAxisRange(-1, 1);
  Joystick.setYAxisRange(-1, 1);
}

int lastStickState[4] = {0,0,0,0};
int lastButtonState[5] = {0,0,0,0,0};

void loop() {

  for (int stickindex = 0; stickindex < 4; stickindex++)
  {
    int currentStickState = !digitalRead(stickindex + 10);
    if (currentStickState != lastStickState[stickindex])
    {
      switch (stickindex) {
        case 0: // UP
          if (currentStickState == 1) {
            Joystick.setYAxis(-1);
          } else {
            Joystick.setYAxis(0);
          }
          break;
        case 1: // RIGHT
          if (currentStickState == 1) {
            Joystick.setXAxis(1);
          } else {
            Joystick.setXAxis(0);
          }
          break;
        case 2: // DOWN
          if (currentStickState == 1) {
            Joystick.setYAxis(1);
          } else {
            Joystick.setYAxis(0);
          }
          break;
        case 3: // LEFT
          if (currentStickState == 1) {
            Joystick.setXAxis(-1);
          } else {
            Joystick.setXAxis(0);
          }
          break;
      }
      lastStickState[stickindex] = currentStickState;
    }
  }

  for (int buttonindex = 0; buttonindex < 5; buttonindex++)
  {
    int currentButtonState = !digitalRead(buttonindex + 3);
    if (currentButtonState != lastButtonState[buttonindex])
    {
      switch (buttonindex) {
        case 0: // FIRE
          Joystick.setButton(0, currentButtonState);
          break;
        case 1: // FIRE 2
          Joystick.setButton(1, currentButtonState);
          break;
        case 2: // FIRE 3
          Joystick.setButton(2, currentButtonState);
          break;
        case 3: // FIRE 4
          Joystick.setButton(3, currentButtonState);
          break;
        case 4: // FIRE 5
          Joystick.setButton(4, currentButtonState);
          break;
      }
      lastButtonState[buttonindex] = currentButtonState;
    }
  }
  delay(10);
}
```

MHeironimus, 2017-05-26:
@jbenam Glad to hear you found a solution.

closed 2017-05-26 by MHeironimus