MHeironimus/ArduinoJoystickLibrary #44: Impossibility to create an 8-axis device

Liben19, 2017-05-05:
(Sorry I don´t speak english this is traslated with Google Tool)
Hi!
I can't create 8 axis joystick.
4 axis don't working.

My sketch:

#include <Joystick.h>
Joystick_ Joystick(
  JOYSTICK_DEFAULT_REPORT_ID,       
  JOYSTICK_TYPE_JOYSTICK,           
  32,                                                 
  2,                                                  
  true,     //Indicates if the X Axis is available on the joystick.
  true,     //Indicates if the Y Axis is available on the joystick.
  true,     //Indicates if the Z Axis (in some situations this is the right X Axis) is available on the joystick.
  true,     //Indicates if the X Axis Rotation is available on the joystick.
  true,     //Indicates if the Y Axis Rotation is available on the joystick.
  true,     //Indicates if the Z Axis Rotation is available on the joystick.
   true,     //Indicates if the Rudder is available on the joystick. //Don't work!
  true,     //Indicates if the Throttle is available on the joystick.
  false,    //Indicates if the Accelerator is available on the joystick. //Don't work!
  false,    //Indicates if the Brake is available on the joystick. //Don't work!
  false);   //Indicates if the Steering is available on the joystick. //Don't work!
void setup() {
   Joystick.begin(); }
void loop() {  }

MHeironimus, 2017-05-08:
@Liben19 
- The Accelerator, Brake, and Steering are turned off in your example above. 
- Can you provide more details regarding your statement on the Rudder (i.e. "Don't work!")? For example, does the host computer see the Rudder? If it does, does the Rudder value change?

Liben19, 2017-05-09:
Hi!
In this case there is no (not determined) axis of the rudder. It turns out the controller on 7 axes.
When using all (11) axes, the axes of the Rudder, Accelerator, Brake and Steering are not defined.

MHeironimus, 2017-08-25:
I have started to look into the issue you describe, and this limitation seems to be a Host OS and/or platform limitation. 

I could not find any axis limitations in the USB HID Usage Tables document ([http://www.usb.org/developers/hidpage/Hut1_12v2.pdf](http://www.usb.org/developers/hidpage/Hut1_12v2.pdf)), however that does not mean they do not exist. In the description of a Multi-axis Controller in Section 4.1 it specifically spells out 6 axes, but that does not mean it is a limit.

Looking at it from a Microsoft Windows perspective, the DirectX API's DirectInput structure for reading in input device information ([https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dideviceobjectinstance.aspx](https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dideviceobjectinstance.aspx)) appears to only support 7 different types of analog axes. The `guidType` member of the `DIDEVICEOBJECTINSTANCE` structure lists the following analog types:

- `GUID_XAxis` : The horizontal axis. For example, it can represent the left-right motion of a mouse.
- `GUID_YAxis` : The vertical axis. For example, it can represent the forward-backward motion of a mouse.
- `GUID_ZAxis` : The z-axis. For example, it can represent the rotation of a mouse wheel or the movement of a joystick throttle control.
- `GUID_RxAxis` : Rotation around the x-axis.
- `GUID_RyAxis` : Rotation around the y-axis.
- `GUID_RzAxis` : Rotation around the z-axis (often a rudder control).
- `GUID_Slider` : A slider axis.

The `DIJOYSTATE` structure ([https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dijoystate.aspx](https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dijoystate.aspx)) seems to support up to 8 axes / analog values: `lX`, `lY`, `lZ`, `lRx`, `lRy`, `lRz`, and two `rglSlider` values. The `DIJOYSTATE2` structure ([https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dijoystate2.aspx](https://msdn.microsoft.com/en-us/library/windows/desktop/microsoft.directx_sdk.reference.dijoystate2.aspx)) supports the same values the `DIJOYSTATE` structure supports, but adds velocity, acceleration, and force versions of the 8 axes / analog values. This implies a limit of either 8 axes (or maybe 8 * 4 axes at the most, if those velocity, acceleration, and force versions could be interpreted as independent axes - but I am not sure that can be done).

Looking at it from a Windows UWP application perspective, you find the following types of input devices ([https://docs.microsoft.com/en-us/uwp/api/windows.gaming.input](https://docs.microsoft.com/en-us/uwp/api/windows.gaming.input)):

- ArcadeStick : Represents an arcade stick.
- FlightStick : Represents a flight stick.
- Gamepad : Represents a gamepad.
- Headset : Contains information about an audio headset attached to a gamepad.
- RacingWheel : Represents a racing wheel.
- RawGameController : [Contains prerelease APIs.] Represents any type of game controller.
- UINavigationController : Represents a controller of any type that can be used for UI navigation. Most gamepads, arcade sticks, and racing wheels can also be accessed as a UINavigationController.

Of these input devices types, the `GamepadReading` stuct ([https://docs.microsoft.com/en-us/uwp/api/windows.gaming.input.gamepadreading](https://docs.microsoft.com/en-us/uwp/api/windows.gaming.input.gamepadreading)) has the most analog values defined, which is 6. There is a new `RawGameController` class ([https://docs.microsoft.com/en-us/uwp/api/windows.gaming.input.rawgamecontroller](https://docs.microsoft.com/en-us/uwp/api/windows.gaming.input.rawgamecontroller)) being added that does not appear to have any limitations, but I am sure there will be some limit to the number of axes an input device could provide.

I am not familar with linux or macOS limitations.

Hope this information helps.

JazzyManSerg, 2017-11-14:
I have similar problem. I make joistick 8 axis, But i see only 7 axiis. Is it bug?

batbsv, 2020-12-17:
win 10 pro 
leonardo boart 8axis set, visible 7.
[image]

project freejoy
https://github.com/FreeJoy-Team/FreeJoy
all 8 axis visible and  usb device visible in dame devices
[image]

how fix this big problem???