MHeironimus/ArduinoJoystickLibrary #45: How to get 16bit precision?

SoulMaril, 2017-05-07:
How to set Arduino Micro to get use of 16bit precision while micro's analog pins can only provide 10bits?

SoylentGraham, 2017-05-07:
Can you be more specific?

Arduino A->D analog input is 10 bit, USB-Joystick protocol outputs at 16bit. 

The code in the repository rescales it for you, depending on how you use it.

Possibly paste some code and someone could confirm your input data is being scaled to 16 bit.

SoulMaril, 2017-05-08:
To be more specific. I am more interested with the input side. I am planning to use a 13bit or 12bit magnetic rotary encoder which has 8192 digits for 360 degrees but Arduino analogRead() can only read 10bits which is 1024 digits.

MHeironimus, 2017-05-08:
@SoulMaril You will need to try one of the following:

- Use an external analog-to-digital converter (e.g. [ADS1115 16-Bit ADC - 4 Channel with Programmable Gain Amplifier](https://www.adafruit.com/product/1085) or equivalent).
- Enhancing ADC resolution by oversampling  (see [http://www.atmel.com/Images/doc8003.pdf](http://www.atmel.com/Images/doc8003.pdf) for more details).

closed 2017-05-12 by MHeironimus