MHeironimus/ArduinoJoystickLibrary #49: It is possible change HID code of a axis?

GuilhermeMoserManerichi, 2017-05-21:
I'm creating a Driving simulator, but there's no clutch at the library. I thought that changing de HID descriptor code at Joystick.cpp would resolve my problem, like change the:

		if (includeRudder == true) {
			// USAGE (Rudder)
			tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
			tempHidReportDescriptor[hidReportDescriptorSize++] = 0xBA;
		}
to:

		if (includeRudder == true) {
			// USAGE (Rudder)
			tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
			tempHidReportDescriptor[hidReportDescriptorSize++] = 0xC6;
		}
But this not work. The axis disappear:
 
[image]

Anybody has some idea how to change it?

Thanks, Guilherme

MHeironimus, 2017-08-24:
I am going to classify this as a feature enhancement request: Add a Clutch to the Joystick library. I will probably try to add a Shifter as well.

mieperDE, 2018-06-18:
Additional axis would be great.