MHeironimus/ArduinoJoystickLibrary #50: Is it possible to have two joysticks without the rotation value?

shingtaklam1324, 2017-07-02:
So I'd like to have a two joystick setup like most game controllers, but the 0deg to 360deg rotation doesn't work for me

kennethnakasone, 2017-07-03:
By "doesn't work for me", do you mean that you do not require the functionality? If so, I believe you can disable them using by setting the 8th, 9th, and 10th parameters of the Joystick constructor to false.

If you meant that "rotation is not being detected", then sorry but I have no idea at the moment.

MHeironimus, 2017-07-04:
You are going to need to provide your sketch file and a description of your hardware setup for the group to be able to help you with your issue.

MHeironimus, 2017-08-24:
I am closing this issue due to a lack of details.

closed 2017-08-24 by MHeironimus