MHeironimus/ArduinoJoystickLibrary #51: Arduino Zero compatiblity ?

Synless, 2017-07-07:
Is it possible to port it so that is can also be compatible with the SAMD21 based board ?
I tried a bit myself but I'm not good enough to do it.
Upon compilation I get "Joystick.h is missing", so I guess that this part of the HID isn't yet written for the Arduino Zero, or that it does not support this part of the HID.

MHeironimus, 2017-07-09:
I have not worked with the Arduino Zero board or any other SAMD21 based boards, so I do not know.

mieperDE, 2018-06-18:
I tried. But I get this error:

`Compiling 'joystickmax' for 'Arduino/Genuino Zero (Programming Port)'
 
Joystick.h:24: In file included from
joystickmax.ino:7: from
 
DynamicHID.h: 37:28: fatal error: PluggableUSB.h: No such file or directory
   #include "PluggableUSB.h"
   compilation terminated

	An error was encountered during the 'Deep Search' library discovery process.
Build failed for project 'joystickmax'
`