MHeironimus/ArduinoJoystickLibrary #54: windows 10 creators edition crash

wimble99, 2017-08-08:
hi I wonder if anyone knows why I can use sliders but as soon as I plug in usb and a button command is in the program not just pressed but used in the code. windows 10 crashes out. Instead of the usual plugged in sound keyboard goes dead (lights out) then a few seconds latter PC dies. btw my program works perfectly on windows 7

MHeironimus, 2017-08-08:
Can you post a copy of your program? Also include which board and IDE you are using.

wimble99, 2017-08-08:
hi IDE 1.8.2 and Leonardo. im not very good at code but it does work fine on my win 7 laptop just tried again this time it hasn't crashed my windows 10 box but the keyboard has stopped working. thanks for taking an interest its much appreciated

#include <Key.h>
#include <Keypad.h>
//-----------------------------------------------------------
const byte ROWS2 = 5; //five rows
const byte COLS2 = 4; //four columns
char keys2[ROWS2][COLS2] = {
  {'1', '2', '3', '4'},
  {'5', '6', '7', '8'},
  {'9', ':', ';', '<'},
  {'=', '>', '?', '@'},
  {'A', 'B', 'C', 'D'}
};

byte rowPins2[ROWS2] = {12, 11, 10, 9, 8}; //connect to the row pinouts of the keypad
byte colPins2[COLS2] = {2, 3, 4, 5}; //connect to the column pinouts of the keypad
Keypad keypad2 = Keypad( makeKeymap(keys2), rowPins2, colPins2, ROWS2, COLS2 );
KeyState getState();

//------------------------------------------------------------

#include "Joystick.h" 

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,
                   0x04  , 20, 0,
                   false, false, false, true, true, true,
                   true, true, false, false, false);

// Set to true to test "Auto Send" mode or false to test "Manual Send" mode.
//const bool testAutoSendMode = true;
const bool testAutoSendMode = false;

// my bit
int slider1 = 1; //pin 1 to slider 1
int slider2 = 2;
int slider3 = 3;
int slider4 = 4;

unsigned int slider1val;
unsigned int slider2val;
unsigned int slider3val;
unsigned int slider4val;
unsigned int k;
unsigned int kp;
unsigned int z[19];
char c;
int lastButtonState[20] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};
unsigned long time= millis();
unsigned long timer;
void setup() {
  // put your setup code here, to run once:
  if (testAutoSendMode)
  {
    Joystick.begin();
  }
  else
  {
    Joystick.begin(false);
  }

  pinMode(A0, INPUT);
  pinMode(A1, INPUT);
  pinMode(A2, INPUT);
  pinMode(A3, INPUT);
  pinMode(A4, INPUT);
  pinMode(A5, INPUT);

  pinMode(13, OUTPUT);
  
keypad2.setDebounceTime(60); 
}

void loop() {
  // put your main code here, to run repeatedly:
  // keys 49=1
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

 
  timer=millis()+50;

do
{
 time=millis();
 
    // statement block
} while (time<timer);

/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
  char key2 = keypad2.getKey();
  
  if (key2 != NO_KEY) {
    // digitalWrite(13, HIGH);
    // delay(1000);
    c = key2;
    c=c-49;
    Joystick.setButton(c,1);
    digitalWrite(13, HIGH);
    

  }

//  keypad2.getState();
if (keypad2.getState()==RELEASED) {
   digitalWrite(13, LOW);
   // for (int i=49; i <= 68; i++){
Joystick.setButton(c,0);
  // Joystick.releaseButton(i-49);
//}
    }
    
 
   

  //------------------------------------------------------
  // my bit
  slider1val =  analogRead(slider1);
  //slider1val = map(slider1val, 0, 1023, 1023, 0);

  slider2val =  analogRead(slider2);
  //slider2val = map(slider2val, 0, 1023, 1023, 0);

  slider3val =  analogRead(slider3);
  //slider3val = map(slider3val, 0, 1023, 1023, 0);
  slider4val =  analogRead(slider4);
  //slider4val = map(slider4val, 0, 1023, 1023, 0);

  Joystick.setThrottle(slider1val);
  Joystick.setRxAxis(slider2val);
  Joystick.setRyAxis(slider3val);
  Joystick.setRzAxis(slider4val);
  /*
    Joystick.setRxAxis(int16_t value)
    Joystick.setRyAxis(int16_t value)
    Joystick.setRzAxis(int16_t value)
    Joystick.setThrottle(int16_t value)
  */

  if (testAutoSendMode == false)
  {
    Joystick.sendState();
  }

}

wimble99, 2017-08-08:
have tried delays and timers up to one second between updates also tried to set it to multi axis but then pc doesn't think its a game controller and it doesn't work

wimble99, 2017-08-08:
update its setting the button number to a value other than zero,,,Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,
                   0x04  , 0, 0,    <<<<<<<<<<<<<<<<<<<<<<<<works sliders work fine

                   false, false, false, true, true, true,
                   true, true, false, false, false);
works but 
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,
                   0x04  , 1, 0,   <<<<<<<<<<<<<<<<<<<<<< or any other number does not it crashes it
                   false, false, false, true, true, true,
                   true, true, false, false, false);

if I use 0x08 its not recognised as anything btw.

wimble99, 2017-08-08:
looks like its solved and was entirely my fault  
false, false, false, true, true, true,
 true, true, false, false, false); 

5 lots of true I changed it to

false, false, false, true, true, true,
 false, true, false, false, false); 
The strange thing it worked fine under windows 7
Thank you for your attention I hope I havnt wasted your time

wimble99, 2017-08-10:
cant get your flight controller test to run on windows 7 either :)

wimble99, 2017-08-10:
can confirm that even example programs crash my win 10 install with clean boot. have done every thing I can to get it to work

wimble99, 2017-08-10:
think ive found it would love to know if anyone cares. you cannot install the up to date arduino drivers because windows 10 knows best and the driver cant handle the leonado as a joystick as far as I can work out. ive been trying to get the true drivers to install but cant make win 10 do it so basically the leonado is useless

wimble99, 2017-08-10:
its usbser.sys btw

wimble99, 2017-08-11:
Please help if possible Im going to have to wipe my hd and install win 7 guess ive found an easy way to shutdown creators edition just plug my arduino in with joystick software :(

MHeironimus, 2017-08-11:
@wimble99 I usually do my library testing on Windows 10, so it should work just fine on Windows 10. I find it interesting that the Leonardo is actually crashing your Windows 10 PC. That makes me think there could be an issue with your hardware (either the Leonardo itself or the USB cable). Are you able to get any of the example programs that come with the library to work?

wimble99, 2017-08-11:
Sorry about my rant its very frustrating i have another leonardo i can try and will change cable also have a friend who will let me try it on his machine will also try reflashing boot loader

wimble99, 2017-08-11:
Hi looks like I was flogging a dead horse the leonardo I was using seems to be the problem hope you forgive me as it works fine in every other way It is quite old I wonder if changes have been made at some point to the firmware any way so far its behaving itself thank you so much for your help I guess I was to close to the problem to try the obvious.

MHeironimus, 2017-08-12:
Glad you were able to get it working.

closed 2017-08-12 by MHeironimus

wimble99, 2017-08-27:
Hi Have found it must be a buffer overflow or something similar as reducing the number of updates has solved the problem just made the controls less sensitive. I make sure an update is only sent when a control is moved, button first pressed and released etc. and also make sure no update can occur without a short delay.  Not to mention windows 10 wont let me use the correct? mother board drivers or the arduino drivers and keeps change back to windows drivers or I get the best drivers are installed message when they clearly are not. Even now if I move the throttle fast and it sends a continuous series of updates it can crash the system. I have many other usb devices including a Saitek flight controller and they all work fine. And of course everything works with windows 7!