MHeironimus/ArduinoJoystickLibrary #61: No Issue Just 2 Questions

pizman, 2017-09-19:
Hi.  Sorry i not found a Email Adress... So i use the issue section to reqeust for help !

First Thank you for this wonderfull code....  I´m pretty new with HID Controllers so i got 2 Questions.

1. Multible HOLD Buttons.
I want to use your System in a Flight Simulator....  There i need lots of NON Momenary Static Switches... that Means it remains PUSHED if it stay in ON Position.
Is THIS a problem with your code when i need for example 15 Buttons on a Controller that ALL Pushed and stay ON same Time ?  In other Words... Is there a limit how many buttons can be ON same time ?

2. Combination of multible Arduinos.
I Normlay work with Arduino Mega for other stuff like showing LED and Displays.....   So i´m a bit confused about the verry low PIn Number on Ardunio Micro/Leonardo.
You say we can work with 32 Buttons, Hat switches and so on....  But i not get enough Pins for all this stuff.
Question:   Is Every Arduino used as a single HID Gamecontroller or is it possible to combine for example 3 Micros so i set 20 Buttons on Controller 1   Hat switches on Controller 2 Axis on Controller 3 ...  BUT All this is show as ONE Gamecontroller in Windows ??
If i understand your readme.....  Is it correct i can define a "ID Number" for the HID Device so i can set for example 3 Arduinos to the SAME HID Gamecontroller ??

Why i ask this.....   The Number of Gamecontroller in my Flightsimulator Setup Area is limmited....  A Proffessional Hardware like LeoBodnar support 32 Buttons, Axis and all the stuff on ONE  Device.
Im just a bit scarred if i would reach the limit pretty fast if i need to define a single GameControlelr Device for every Arduino if i only have so less pins.

Thank you !    And again sorry for posting this in wrong area !

pizman, 2017-09-20:
Dear Matt.
At Firt THANK YOU for your incredible support.... A Reply to a question within some hours is wunderfull !

I not be advanced enough to use shift registers or to implement Matrix in the script myself ( i just beginning with the Arduino scripting) .....

So my last question is simple....

Do you see a way in the future to support YOUR System for a Arduino Board with more Pins ( Like the MEGA)  ?
There we got already  52 IO Pins (Enogh for 32 Buttons) and a lot of Analog Pins to set the Axis...... So a Using of Registers and Matrix is not needed.

I Read in the TXT, that the Mega Not be able to work with your Code.....  And i fully understand for your personal use the low number of Pins are Ok ( or you work with Matrix /Shift Registers)

But finaly for the "dummy" User like me....  Is it possible to make a MEGA to a Full 32 Button Game Controller with axis ??  Can you help us there ??

( I know it´s unfriendly to ask you for giving me a recommendation to a other software..... But maby you can help me  )

Thanks again for your work, Sir !
Dear Matt.
At Firt THANK YOU for your incredible support.... A Reply to a question within some hours is wunderfull !

I not be advanced enough to use shift registers or to implement Matrix in the script myself ( i just beginning with the Arduino scripting) .....

So my last question is simple....

Do you see a way in the future to support YOUR System for a Arduino Board with more Pins ( Like the MEGA)  ?
There we got already  52 IO Pins (Enogh for 32 Buttons) and a lot of Analog Pins to set the Axis...... So a Using of Registers and Matrix is not needed.

I Read in the TXT, that the Mega Not be able to work with your Code.....  And i fully understand for your personal use the low number of Pins are Ok ( or you work with Matrix /Shift Registers)

But finaly for the "dummy" User like me....  Is it possible to make a MEGA to a Full 32 Button Game Controller with axis ??  Can you help us there ??

( I know it´s unfriendly to ask you for giving me a recommendation to a other software..... But maby you can help me  )

Thanks again for your work, Sir !

MHeironimus, 2017-09-21:
@pizman The key to this library's flexibility and ease of use is the fact that the USB communications are done via the main ATmega32u4 microcontroller, rather than some other piece of hardware (e.g. the FTDI FT232RL for the Arduino Mega you reference). This allows the library to make the USB device look like anything you want, rather than being "hard-coded" as a serial port (I know this is not technically true, but...). 

I do not currently support this (as I have not had a chance to try it out), but a fellow contributor, @sreksten, has written up some instructions on how to use the Mega and the older version of the library. These instructions can be found at Issue https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/37 .

Hope these comments help,
Matt

closed 2017-10-20 by MHeironimus