MHeironimus/ArduinoJoystickLibrary #65: Games won't recognice my device

legomolina, 2017-10-08:


MHeironimus, 2018-02-24:
Are you saying the Windows game controller test dialog does not display any of your button presses? If so, I would recommend writing the value of one of the pins out to the Serial Port so you can verify the pins are getting set/pressed the way you expect. I usually put the pins in `INPUT_PULLUP` mode and have the buttons connected to ground.

legomolina, 2018-02-25:
No, Windows game controller dialog works perfect. But when I execute a game and go to options > controls, I can't bind buttons from the arduino to actions.

kingfisher09, 2020-05-09:
Did you find a solution to this issue?

legomolina, 2020-05-09:
No. Just some games allow to interact with Arduino and others no. I'm not sure if it's a problem with this lib or maybe it's a game/Windows issue.

kingfisher09, 2020-05-09:
I feel like the solution is to mimic the number of inputs of a standard controller. As shown below. 

[image]

Rather than like this:
[image]

I think the key to this is editing the Joystick files in the library. I managed to remove the Rudder and Throttle by commenting out parts of the files but unfortunately that stopped the rest of the functionality too. Any ideas?

targodan, 2020-05-09:
[As per the README](https://github.com/MHeironimus/ArduinoJoystickLibrary#joystick_) you can specify the number of axis and buttons in the constructor.

kingfisher09, 2020-05-09:
Haha, yes sorry I just noticed that!

closed 2022-03-30 by MHeironimus