MHeironimus/ArduinoJoystickLibrary #67: Second joystick not recognized by OS

faxm0dem, 2017-11-27:
Thanks for this awesome library!
I managed to run it smoothely when only defining one joystick.
When I setup a second one, the OS doesn't see it: it only registers `/dev/input/js0` and no `js1`.
Here's the minimal test code:

```
#include <Joystick.h>

Joystick_ Joy1(0x03, JOYSTICK_TYPE_JOYSTICK, 2, 0, true, true, false, false, false, false, false, false, false, false, false);
Joystick_ Joy2(0x04, JOYSTICK_TYPE_JOYSTICK, 2, 0, true, true, false, false, false, false, false, false, false, false, false);

void setup() {
  Joy1.begin();
  Joy2.begin();
}
```

and dmesg output:

```
[ 7397.746824] input: Arduino LLC Arduino Micro as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:2341:8037.0011/input/input32
[ 7397.747064] hid-generic 0003:2341:8037.0011: input,hidraw2: USB HID v1.01 Joystick [Arduino LLC Arduino Micro] on usb-0000:00:14.0-3/input2
```

MHeironimus, 2017-11-28:
Take a look at the following article and see if it fixes your issue:
http://mheironimus.blogspot.com/2015/09/linux-support-for-arduino-leonardo.html

Please indicate if this fix works for you. If not, please post what OS and version you are using.

faxm0dem, 2017-11-28:
Yes it does work! Apologies for being too lazy to search through closed issues.
I'm repenting by proposing an addition to the documentation and code comments!

MHeironimus, 2017-12-21:
Updated FAQ to document how to resolve this issue.

closed 2017-12-21 by MHeironimus