MHeironimus/ArduinoJoystickLibrary #69: No matching Function for Call to 'Joystick_::Joystick_(int, int, int, int, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool)'

Crashdummyy, 2017-12-05:
Hey There 
I am currently stuck.
I'm getting this error:
`Arduino: 1.6.7 (Linux), Board: "Arduino Leonardo"

In file included from /media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/Arduino.h:225:0,
                 from sketch/PiGameboy_Me.ino.cpp:1:
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/USBAPI.h:31:2: error: 'ring_buffer' does not name a type
  ring_buffer *_cdc_rx_buffer;
  ^
In file included from /media/crashdummy/Festplattiii/Download/arduino-1.6.7/libraries/Joystick/src/DynamicHID/DynamicHID.h:27:0,
                 from /media/crashdummy/Festplattiii/Download/arduino-1.6.7/libraries/Joystick/src/Joystick.h:24,
                 from /media/crashdummy/Festplattiii/Linux_DL/PiGameboy_Me/PiGameboy_Me.ino:5:
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/PluggableUSB.h:35:22: error: 'USBSetup' has not been declared
   virtual bool setup(USBSetup& setup) = 0;
                      ^
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/PluggableUSB.h:37:29: error: 'USBSetup' has not been declared
   virtual int getDescriptor(USBSetup& setup) = 0;
                             ^
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/PluggableUSB.h:57:21: error: 'USBSetup' has not been declared
   int getDescriptor(USBSetup& setup);
                     ^
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/PluggableUSB.h:58:14: error: 'USBSetup' has not been declared
   bool setup(USBSetup& setup);
              ^
In file included from /media/crashdummy/Festplattiii/Download/arduino-1.6.7/libraries/Joystick/src/Joystick.h:24:0,
                 from /media/crashdummy/Festplattiii/Linux_DL/PiGameboy_Me/PiGameboy_Me.ino:5:
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/libraries/Joystick/src/DynamicHID/DynamicHID.h:80:3: error: 'InterfaceDescriptor' does not name a type
   InterfaceDescriptor hid;
   ^
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/libraries/Joystick/src/DynamicHID/DynamicHID.h:82:3: error: 'EndpointDescriptor' does not name a type
   EndpointDescriptor  in;
   ^
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/libraries/Joystick/src/DynamicHID/DynamicHID.h:106:21: error: 'USBSetup' has not been declared
   int getDescriptor(USBSetup& setup);
                     ^
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/libraries/Joystick/src/DynamicHID/DynamicHID.h:107:14: error: 'USBSetup' has not been declared
   bool setup(USBSetup& setup);
              ^
In file included from /media/crashdummy/Festplattiii/Linux_DL/PiGameboy_Me/PiGameboy_Me.ino:5:0:
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/libraries/Joystick/src/Joystick.h:58:7: error: redefinition of 'class Joystick_'
 class Joystick_
       ^
In file included from /media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/Arduino.h:225:0,
                 from sketch/PiGameboy_Me.ino.cpp:1:
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/USBAPI.h:73:7: error: previous definition of 'class Joystick_'
 class Joystick_
       ^
PiGameboy_Me:18: error: no matching function for call to 'Joystick_::Joystick_(int, int, int, int, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool)'
   false, false, false);  // No accelerator, brake, or steering
                      ^
/media/crashdummy/Festplattiii/Linux_DL/PiGameboy_Me/PiGameboy_Me.ino:18:22: note: candidates are:
In file included from /media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/Arduino.h:225:0,
                 from sketch/PiGameboy_Me.ino.cpp:1:
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/USBAPI.h:76:2: note: Joystick_::Joystick_()
  Joystick_();
  ^
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/USBAPI.h:76:2: note:   candidate expects 0 arguments, 15 provided
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/USBAPI.h:73:7: note: constexpr Joystick_::Joystick_(const Joystick_&)
 class Joystick_
       ^
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/USBAPI.h:73:7: note:   candidate expects 1 argument, 15 provided
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/USBAPI.h:73:7: note: constexpr Joystick_::Joystick_(Joystick_&&)
/media/crashdummy/Festplattiii/Download/arduino-1.6.7/hardware/arduino/avr/cores/arduino/USBAPI.h:73:7: note:   candidate expects 1 argument, 15 provided
/media/crashdummy/Festplattiii/Linux_DL/PiGameboy_Me/PiGameboy_Me.ino: In function 'void tAnalogSticks()':
PiGameboy_Me:195: error: 'class Joystick_' has no member named 'setXAxis'
   Joystick.setXAxis(map(x1Final, x1F1, x1F2, 0, 255));
            ^
PiGameboy_Me:196: error: 'class Joystick_' has no member named 'setYAxis'
   Joystick.setYAxis(map(y1Final, y1F1, y1F2, 0, 255));
[... 103 lines cut ...]
   Joystick.setRxAxisRange(-127, 127);
            ^
PiGameboy_Me:326: error: 'class Joystick_' has no member named 'setXAxis'
   Joystick.setXAxis(0);
            ^
PiGameboy_Me:327: error: 'class Joystick_' has no member named 'setYAxis'
   Joystick.setYAxis(0);
            ^
PiGameboy_Me:328: error: 'class Joystick_' has no member named 'setRxAxis'
   Joystick.setRxAxis(0);
            ^
PiGameboy_Me:329: error: 'class Joystick_' has no member named 'setRyAxis'
   Joystick.setRyAxis(0);
            ^
PiGameboy_Me:331: error: 'class Joystick_' has no member named 'begin'
   Joystick.begin();
            ^
exit status 1
no matching function for call to 'Joystick_::Joystick_(int, int, int, int, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool)'
`

To Be honest that's the best I managed so far.
I did load v1.6.7 to get it to work to this Point.

I want to use the Pro Micro as Gameboy Gamepad ( 13 Buttons + 1 PSP Analog stick)

MHeironimus, 2017-12-05:
Are you able to compile and use some of the example sketches that are included with the library?

Crashdummyy, 2017-12-06:
Can't do that :(
It's everytime the same error Message :/

MHeironimus, 2017-12-06:
I would try reinstalling the Arduino IDE (maybe upgrade to the latest version) and reinstall the latest version of the Arduino Joystick Library (2.0.4 when I wrote this) and see if you are still having the issue. The fact that even the simple sketch files are not working, makes me think it could be something with the Arduino IDE.

Do the Keyboard or Mouse examples that come with the Arduino IDE work for you?

Crashdummyy, 2017-12-06:
Thanks for the Quick Reply.
I will try that this evening :)

Crashdummyy, 2017-12-06:
Did a fresh install now I have dependency problems :/
Just hatin the arduino IDE...

The joystick folder is inside every library folder.
I can find it in the context menue as well.
Yet it won't find Joystick.h....
If I move it out of the ./src Folder it won't find another dll.....

Crashdummyy, 2017-12-07:
The Example works with portable 1.8.5 on windows ( just added into %path%/libraries ).
With the installed version I get the same error :laughing: 

So I guess this issue will be closed, because from here on every Issue will now have to do with my code ;)

closed 2017-12-07 by Crashdummyy