MHeironimus/ArduinoJoystickLibrary #73: Compiler Error on Arduino 1.6.6 for Arduino Due

MHeironimus, 2017-12-27:
The following compiler error occurs on the Arduino 1.6.6 if the target board is the Arduino Due:

~~~
C:\Users\heiro\Documents\Arduino\libraries\Joystick\src\Joystick.cpp: In member function 'int Joystick_::buildAndSet16BitValue(bool, int16_t, int16_t, int16_t, int16_t, int16_t, uint8_t*)':

C:\Users\heiro\Documents\Arduino\libraries\Joystick\src\Joystick.cpp:592:54: error: 'max' was not declared in this scope

  int16_t realMaximum = max(valueMinimum, valueMaximum);

                                                      ^

exit status 1
Error compiling.
~~~

Mrtulusoy, 2017-12-27:
3 days I can not sleep please help

BleuLlama, 2018-01-16:
This should patch it;  in the top of C:\Users\heiro\Documents\Arduino\libraries\Joystick\src\Joystick.cp, just after the line "#include "Joystick.h"", add this:
```
#ifndef max
#define max( A, B ) \
((A)>(B))?(A):(B)
#endif
```

MHeironimus, 2018-02-24:
It turns out it is not the Arduino IDE 1.6.6 that is the issue, but it is the version of the **Arduino SAM Boards (32-bits ARM Cortext-M3)** that was causing the issue. You must use version 1.6.7 or above.

closed 2018-02-24 by MHeironimus