MHeironimus/ArduinoJoystickLibrary #74: Z-Axis not being incremented properly - Fully Off to Fully On

aksuited, 2017-12-30:
I am sorry to post this, but I am not sure of the cause as I have tested the library with examples and my own code and I have no issue using the buttons or the X & Y axis, but the Z-axis functionality is not at all working how I expect, even in the example joystick test.

Specifically the Z-axis in the joystick text example provided in the library is going from fully on, to fully off.   Using the windows utility and another input utility it appears the Z-axis is going to fully on with any value sent to it above the mid point of the libraries set scale, and to fully off with any value below the mid point of the scale.

I.E.:
  Joystick.setZAxisRange(0, 1000);

  Joystick.setZAxis(500);   //Joystick reports to windows it is at it's mid or centered point on the Z-axis with a raw value of 32767

 Joystick.setZAxis(499);   //Joystick reports to windows it is at it's 0 point on the Z-axis with a raw value of 0

 Joystick.setZAxis(501);   //Joystick reports to windows it is at it's 0 point on the Z-axis with a raw value of 65535

rar0n, 2018-03-13:
You might need to go through the controller calibration in Windows to register the range it has on all axises. Not sure if this is your problem, but I had this issue a while back (not with this library).

MHeironimus, 2020-06-25:
Could not reproduce issue.

closed 2020-06-25 by MHeironimus