MHeironimus/ArduinoJoystickLibrary #76: Any way to detect whether joystick is using in Windows?

damul, 2018-02-02:
Is there any method to detect joystick is using in Windows?

In my application, I am using 2 different modes, one is joystick mode, and the other is not. 
In both modes, I update joystick values, but I want to know these values are used or not, 
to show led indicator current mode. 

The default is "not joystick" mode, but joystick values are updated, too. 
When values are used in windows, Arduino goes to "joystick" mode, to change led indicator.

Is it possible scenario?

MHeironimus, 2018-02-02:
Is your question specific to Microsoft Windows or do you just want to know if the device is connected to a computer?

In other words, which of the following best describes your situation:

1. When the device is connected a Linux machine (or Mac or something else), you are in mode "A", but if you connect to a Microsoft Windows machine, you are in mode "B".
2. When the device is not connected to a host computer, you are in mode "A", but if you are connected to a host computer, you are in mode "B".

If you are looking for a solution to option 1, I really do not know. I am not sure how to make a USB device aware of what OS it is connected to. I am not even sure that is possible (it may be, I just don't know).

If you are looking for a solution to option 2, there may be a way to detect when the USB connection is made. I would have to do more research on this.

damul, 2018-02-08:
Thank you for your answer. 

In my case, it is similar to option 2. 
However, it is slightly different. I actually want to know that the device is 'actually' used or not. 
It means when you just connect the device to PC, it remains in mode "A" 
until specific software that uses the joystick data launches.

SukkoPera, 2020-04-07:
Any news on this? I have noticed that SendState() is slow to return when no application on the host is reading the joystick, so I'd like to avoid calling it in this case.