MHeironimus/ArduinoJoystickLibrary #82: Library give error when compiling

Dinges64, 2018-02-25:
## Description of Issue
Enter description of issue here.

C:\Users\peggy\AppData\Local\Temp\cczoa4XP.ltrans0.ltrans.o: In function `Joystick_::sendState() [clone .constprop.30]':

F:\Gebruikers\peggy\Documents\Arduino\libraries\Joystick/Joystick.cpp:677: undefined reference to `DynamicHID()'

F:\Gebruikers\peggy\Documents\Arduino\libraries\Joystick/Joystick.cpp:677: undefined reference to `DynamicHID_::SendReport(unsigned char, void const*, int)'

C:\Users\peggy\AppData\Local\Temp\cczoa4XP.ltrans1.ltrans.o: In function `__base_ctor ':

F:\Gebruikers\peggy\Documents\Arduino\libraries\Joystick/Joystick.cpp:439: undefined reference to `DynamicHID()'

F:\Gebruikers\peggy\Documents\Arduino\libraries\Joystick/Joystick.cpp:439: undefined reference to `DynamicHID_::AppendDescriptor(DynamicHIDSubDescriptor*)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Fout bij het compileren van board Arduino Leonardo
## Technical Details

- Arduino Board (e.g. Arduino Leonardo): 
  Arduino Leonardo
- Host OS (e.g. Windows 10):
Windows 10
- Arduino IDE Version (e.g. 1.8.3):
1.8.1
## Sketch File that Reproduces Issue
// Simple example application that shows how to read four Arduino
// digital pins and map them to the USB Joystick library.
//
// The digital pins 9, 10, 11, and 12 are grounded when they are pressed.
//
// NOTE: This sketch file is for use with Arduino Leonardo and
//       Arduino Micro only.
//
// by Matthew Heironimus
// 2015-11-20
//--------------------------------------------------------------------

#include <Joystick.h>
// Create the Joystick
/*Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
                   JOYSTICK_TYPE_MULTI_AXIS, 4, 0,
                   false, false, false, false, false, false,
                   false, false, true, true, true);
*/
Joystick_ Joystick;
void setup() {

  // Initialize Button Pins
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
  pinMode(6, INPUT_PULLUP);
  pinMode(7, INPUT_PULLUP);
  pinMode(8, INPUT_PULLUP);
  pinMode(9, INPUT_PULLUP);
  pinMode(10, INPUT_PULLUP);
  pinMode(11, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  Serial.begin(9600);
  // Initialize Joystick Library
  Joystick.setAcceleratorRange(0, 1023);
  Joystick.setBrakeRange(363, 1023);
  Joystick.setSteeringRange(400, 600);
  Joystick.begin();
}

// Constant that maps the phyical pin to the joystick button.
const int pinToButtonMap = 2;

// Last state of the button
int lastButtonState[11] = {0,0,0,0,0,0,0,0,0,0,0};

void loop() {

  // Read pin values
  for (int index = 0; index < 11; index++)
  {
    int currentButtonState = !digitalRead(index + pinToButtonMap);
    if (currentButtonState != lastButtonState[index])
    {
      Joystick.setButton(index, currentButtonState);
      lastButtonState[index] = currentButtonState;
    }
  }
  int steeringwheel = 1023-analogRead(A0);
  int BrakePot = 1023-analogRead(A3);
  int ThrottlePot = 1023-analogRead(A4);
  Joystick.setSteering(steeringwheel);
  Joystick.setThrottle(ThrottlePot);
  Joystick.setBrake(BrakePot);
  Serial.println("Stuur = ");
  Serial.println(steeringwheel);
  Serial.println("Gas = ");
  Serial.println(ThrottlePot);
  Serial.println("Rem = ");
  Serial.println(BrakePot);
  
  delay(25);
}

~~~
Sketch file goes here (if applicable)
~~~

## Wiring Details

Any details about pin wirings that may be relevant.

MHeironimus, 2018-02-26:
I do not see any errors when I try to compile your sketch:

~~~
Sketch uses 9200 bytes (32%) of program storage space. Maximum is 28672 bytes.
Global variables use 324 bytes (12%) of dynamic memory, leaving 2236 bytes for local variables. Maximum is 2560 bytes.
~~~

Dinges64, 2018-03-05:
Thanks for your reaction
In my I have to add #include <DynamicHID.h> before it compiles without errors.
After this in Windows 10 it is not seen as a game device when i choose JOYSTICK_TYPE_MULTI_AXIS.
Do you know how to solve this

MHeironimus, 2018-06-13:
Take a look at Issue #88 and see if that addresses your question regarding the device type.

closed 2018-06-13 by MHeironimus