MHeironimus/ArduinoJoystickLibrary #83: Arduino pmicro Joystick sketch doesnt appear as a "game Controller"

Ravny77, 2018-03-28:
## Description of Issue
I've been following a tutorial to make an analog handbrake for car simulation games and its been going fine until i've reached the stage to configure the controller/joystick. The arduino does not appear in "Sound, Video and Gaming Controllers" instead it appears in "Ports (COM & LPT)" under the name "Arduino Micro (COM3). 
I have tried many things, i've spent about a combined total of 4 hours or more just troubleshooting and trying different things from different forums and people, but with no solution.

## Technical Details

I'm running on Windows 10, i've heard this could be a cause of some issues.
I have a Arduino Pro Micro board
other than that im very new to this sort of stuff, infact this is the only experience ive had ever.

## Sketch File that Reproduces Issue

no issues come up with the sketch file, the only issue im having is configuring the controller/joystick.

any help i am very appreciative of, im very stuck and have put quite a bit of time and effort to get to this very last stage xD

MAG0101, 2018-07-02:
I have the same on Windows 10 machines. I tried a Joystick Test Software, and its working without being displayed. When you open the device manager it appears under HID-Devices, same as on a Windows 7 machine.

The Arduino IDE seems to do something in the background I can not retrace.

I put a working example from the Joystick Lib on an Arduino Micro, and its working. Putting this controller into a fresh installed Windows 7 or 10 Machine also shows up in the device manager as described. But the Joystick Test Software doesn't find any Joystick. Only when you install the Arduino IDE on that machine and flash an Joystick example to a 32U4 controller (e.g.Arduino Micro/Sparkfun Pro Micro) all other controllers plugged into this PC are working without problems. Maybe anybody has an explanation for that behavior.

Joystick Test Software used: http://www.mcrenox.com.ar/downloads/joytester.exe

SunnySundays, 2019-07-15:
I have this issue as well, my joystick is recognized as a keyboard. Please if someone knows how to change device to game controller would be very helpful.

crstrand, 2020-11-24:
If you open joy.cpl does it show there?
[image]

avgeeklucky, 2020-12-04:
Related: #83
I'm having the same issue. I halve not looked into too much, but too me it seems windows is getting confused with multiple devices on the Arduino USB device:
[image]
For me its not showing up in Game Controllers:
[image]

Games do recognise it a joystick. I've used [Universal Control Remapper](https://github.com/Snoothy/UCR) to get some output about buttons pressed and axes position.

bvandevliet, 2021-03-23:
Check out this post, it fixed the problem for me: [https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/131#issuecomment-778864974](url)

schnoog, 2021-05-08:
> Check out this post, it fixed the problem for me:
> 
> _Originally posted by @zebolc in [#131 (comment)](https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/131#issuecomment-778864974)_

Could you please provide some details about the driver which fixed your problem.

Created a controller which worked quite well but had to reinstall Windows 10. Now it's no longer recognized as game controller.

Thanks in advance

zebolc, 2021-05-13:
Hi,

I did not install any driver, I just changed this parameter in
constructor:  JOYSTICK_TYPE_MULTI_AXIS into JOYSTICK_TYPE_JOYSTIC, and that
worked for me.

BR,
Natalia