MHeironimus/ArduinoJoystickLibrary #96: Bluetooth support

legomolina, 2018-09-15:
**Description**
Is possible to use this library with any Bluetooth module for Arduino?

Oky12, 2018-12-23:
if you want to make a wireless controller with bluetooth. You need a wireless HID like RN 42 (Bluetooth HID)

PascalLeroi, 2019-04-12:
I use a Feather 32u4 Bluefruit LE and after a lot of trying I stumbled across this library, which makes it work awesome via USB, but I would indeed like to know how to use it over bluetooth :) What changes do I need to make and such. Thanks in advance.

rndsumgen, 2019-04-12:
Betting it suffers from the problem I was having with the Feather M4 express which is it cuts off the SPI pin Not sure if it's changing clock or buss communication but betting it's one of the two interesting it does it on a different chip.

hasbeeen, 2021-01-16:
https://github.com/hasbeeen/arduino-bt-game-steering-wheel-.git   I added bluetooth and jayroscobe with a simple coding, I think it works fine.