MHeironimus/ArduinoJoystickLibrary #99: Support added for Arduino Zero, M0, MKR family

gdsports, 2018-10-31:
https://github.com/gdsports/ArduinoJoystickLibrary/tree/samd_patch

I added support for SAMD21 boards (Zero/M0 and MKR family). In addition, the changes work with the Adafruit Metro M4 board (SAMD51). There are so many SAMD boards from Arduino, Adafruit, and Sparkfun but I can only test with the small number of boards I have. In theory, all Arduino compatible SAMD21 boards should work but a lot depends on how closely the compatible boards track ArduinoCore-samd.

Tested: Arduino Zero, Arduino MKR Zero, and Adafruit Metro M4. The Adafruit board is based on the Microchip SAMD51 but it works.

This change does not use HID.h so it is different from PR #90. The change is patterned on the Arduino Due change.

It would be nice for someone to try the changes on my fork (note the changes are on a branch named samd_patch, not master) and report success or failure. Or I can submit a PR.

ransombot, 2019-04-11:
Thank you, using the 1 button sample on my M4 feather right now.  Will test more later with axies and report if needed.

ransombot, 2019-04-12:
Appears to disable SPI communication for me, disabling wings & other devices like gyro's is this expected?

ttait-vantim, 2020-11-13:
I tried compiling JoystickTest for the SAMD21 based Adafruit ItsyBity M0  (IDE 1.8.12), but I get this error:

Arduino\libraries\ArduinoJoystickLibrary-master\src/DynamicHID/DynamicHID.h:37:12: fatal error: PluggableUSB.h: No such file 
or directory
   37 |   #include "PluggableUSB.h"

It compiles fine if I target the ItsyBitsy 32u4.

Any ideas?

Thanks-

Tim

antoniomolram, 2021-03-14:
I am having exactly the same issue as @ttait-vantim, did you figure it out @ttait-vantim?

ttait-vantim, 2021-03-15:
No, but I didn't try too hard.  I switched to a Itsy-Bitsy 32u4. 

Tim

jamesy0ung, 2021-11-09:
Same issue with ItsyBitsy M0

LeboisVR, 2022-03-19:
doesn't work on samd21 mini :
`USB/PluggableUSB.h: No such file or directory
   #include "USB/PluggableUSB.h"
            ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
`

andreasdahl1987, 2022-04-04:
> doesn't work on samd21 mini : `USB/PluggableUSB.h: No such file or directory #include "USB/PluggableUSB.h" ^~~~~~~~~~~~~~~~~~~~ compilation terminated. `

You'll have to roll back the core driver to at least 1.6.19. That worked for me

hapklaar, 2022-07-06:
> https://github.com/gdsports/ArduinoJoystickLibrary/tree/samd_patch
>
> I added support for SAMD21 boards (Zero/M0 and MKR family). In addition, the changes work with the Adafruit Metro M4 board [...]

I applied your patch to the current source and that fixed the compilation errors for my Seeeduino XIAO. Thanks!