MHeironimus/ArduinoJoystickLibrary #100: More than 32 buttons and 2 hats

pw-sys, 2018-11-02:
**Describe your feature request**
Is it possible to change the maximum number of hats to 4 or 5  and the maximum number of buttons to 64 or 128?

MHeironimus, 2020-06-09:
These tend to be limitations of the software that is integrating with the device. For example:

> XInput supports maximum of 4 axes, 10 buttons, 2 triggers and 8-direction digital pad per controller, compared to DirectInput's support for 8 axes, 128 buttons, and full-range POV. (The number of axes, buttons and triggers XInput supports corresponds directly to the Xbox 360 controller.)

From https://en.wikipedia.org/wiki/DirectInput

I know you can increase the button count beyond 32 (see https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/FAQ), but the program you are using it with may or may not support them.

I have tried to do 4 hats in the past, but could not get it to work consistently enough to officially add it to the library. I can add that as a future feature request.

mauriciokami, 2020-06-30:
MHeironimus, could you please make a code to accept more than 32 buttons (64 buttons)?
because I'm using shift register but the code doesn't accept more than 32 buttons.
Thank you

MHeironimus, 2020-06-30:
@mauriciokami The library currently supports over 32 buttons (32 is just the default value). The default Windows Joystick test application can only display 32 buttons. You have to use an application that supports over 32 buttons to see them. For example:

[image]

Here is a quick sketch that defines 64 buttons (it only sets the last 4 buttons for simplicity).

```C++
#include <Joystick.h>

Joystick_ Joystick(0x06, JOYSTICK_TYPE_GAMEPAD, 64, 0, 
  false, false, false, false, false, false, false, false, false, false, false);

void setup() {
  // Initialize Button Pins
  pinMode(9, INPUT_PULLUP);
  pinMode(10, INPUT_PULLUP);
  pinMode(11, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin();
}

// Constant that maps the phyical pin to the joystick button.
const int pinToButtonMap = 9;

// Last state of the button
int lastButtonState[4] = {0,0,0,0};

void loop() {

  // Read pin values
  for (int index = 0; index < 4; index++)
  {
    int currentButtonState = !digitalRead(index + pinToButtonMap);
    if (currentButtonState != lastButtonState[index])
    {
      Joystick.setButton(63 - index, currentButtonState);
      lastButtonState[index] = currentButtonState;
    }
  }

  delay(50);
}

```

MHeironimus, 2020-06-30:
@mauriciokami FYI: 128 buttons also works.

[image]

HUANGJIN169, 2020-07-01:
very good

Jack-XHP, 2020-07-09:
Is it possible to create 3 instances of joystick to solve the 32 buttons limit? Although some games handle 128 buttons without problem, i.e. DCS. But a lot of games still ignore anything over 32 buttons. I read some info from another Arduino HID project, and the dev mentioned that 32u4 is capable to handle 3 usb device using single report and can expend to more if multi report is used.

oneandrelom, 2020-07-26:
Is it possible to have more than 32 buttons (48 would be nice) and 6 Axis? I tried assigning the flags on the gamepad to true,  with no success.

MHeironimus, 2020-07-26:
@oneandrelom  Yes, but the software you are using must support it. For example, the built-in Windows Test application only supports 32.  See https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/FAQ for more details.

oneandrelom, 2020-07-26:
@MHeironimus Thanks, the example, MultipleJoystickTest, provided the solution. I configured a gamepad as a 32 button and a joystick as a 32 button, six axis device.