MHeironimus/ArduinoJoystickLibrary #102: handbrake not recognized

iPrazeres, 2018-11-08:
I have created a handbrake using a potetiometer connected to pin A0, and I have 6 buttons from pin 2-7. the games recognize my buttons but not the potentiometer, i can calibrate it on windons but it still doesn't recognize it. using this site"http://html5gamepad.com/" i see that it never wakes up the 0AXIS but the value change when i move the hand brake
[](https://imgur.com/a/d1BLZaH)
if you see the image my wheel has the three active when i press the pedals.
everything works in games apart from the hand brake that the games don't recognize it

Here is he code
`#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_GAMEPAD, 6, 0,// Button Count, Hat Switch Count
  false, false, false, //X,Y,Z
  false, false, false, //Rx,Ry,Rz
  false, false, // rudder, throttle
  false, true, false); // accelerator, Brake , Steering

void setup() {
 //handbrake
  pinMode(A0, INPUT); 
  // Initialize Button Pins
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
  pinMode(6, INPUT_PULLUP);
  pinMode(7, INPUT_PULLUP);
  // Initialize Joystick Library
  
  
  Joystick.begin();
 
  
  }

// Constant that maps the phyical pin to the joystick button.

const int pinToButtonMap = 2;

// Last state of the button
int lastButtonState[6] = {0,0,0,0,0,0};

void loop() {

  // Read pin values
  for (int index = 0; index < 6; index++)
  {
    int currentButtonState = !digitalRead(index + pinToButtonMap);
    if (currentButtonState != lastButtonState[index])
    {
      Joystick.setButton(index, currentButtonState);
      lastButtonState[index] = currentButtonState;
    }
  }

  delay(50);
  
  {int pot = analogRead(A0);

int mapped = map(pot,0,1023,-14,255);
{Joystick.setBrake(mapped);}}
}`

iPrazeres, 2018-11-08:
if i use the version 1.0 of your joystick it works cause it has like 5AXIS activated, but with that one i can't reduce the number of axis or buttons

vikhattangady, 2019-08-16:
If you are still having this problem, 1 thing you could try is to map the handbrake to the x-axis instead of the brake axis. It may be possible that some games do not recognise so many axes. All should recognise x-axis.
Then in the game settings, you can set that (x-axis) to be the handbrake 'key" / axis