MHeironimus/ArduinoJoystickLibrary #104: Defining which range of values will be reported to the HID interface

kharar, 2018-11-25:
Hi, great library, seems to work pretty consistently on Windows 7 and 10, great job!!

The flight simulator (FPV Freerider) that I am using is unable to recognize joystick positions in the negative region, that is having the center point at zero.
Here is their developers explanation: "On Windows, you need to setup so that the joystick channels go from 0 at minimum to 100 at maximum
(instead of -100 to +100)."

I have been fiddling with setting the AxisRange() but that does not seem to help a lot.
What I would really like is a procedure to set what max- and minimum values will be reported to the HID interface.
Is there any way that I could change this behavior within Arduino myself?
....or did I completely misunderstand the meaning of the AxisRange() procedures?

kharar, 2018-11-26:
Well, just wanted to let you know that eventually I made it work with FreeriderFPV

I was able to fix my problem by changing the -_MINIMUM values within the joystick.cpp file like this:

```
#define JOYSTICK_AXIS_MINIMUM 0 // was -32767
#define JOYSTICK_AXIS_MAXIMUM 32767
#define JOYSTICK_SIMULATOR_MINIMUM 0 // was -32767
#define JOYSTICK_SIMULATOR_MAXIMUM 32767
```
Considering this I guess it would be pretty easy to construct a procedure call to adjust the max/min values

PsykomantaBrain, 2018-12-20:
Turns out this applies to all Unity-based games. I was having the same problems here in my project and it was driving me nuts.

Thanks for posting the solution!

Cheers

spaelectronics, 2019-01-26:
I was also having this problem, and the suggested settings above worked!

gregnau, 2019-03-08:
Actually this hack is decreasing the resolution to half on the axes and hat switches.
I have made a proper 16 bit unsigned version, which is sending values between 0-65535 (instead of 32767):
[https://github.com/gregnau/ArduinoJoystickLibrary](https://github.com/gregnau/ArduinoJoystickLibrary)

Working nice with Unity, but idk how it plays with other games.

gregnau, 2019-03-31:
After a few weeks of experimenting with this unsigned mod, it seems like maybe not the best solution. Worked like a charm under Windows 10 (1803) and OSX, but had serious issues on Ubuntu (16.04).
It is expecting values between -32767 and 32767, while the proper range is transmitter with the usb descriptor.
Modifying the range used in the library to 0-32767 and setting the descriptors according solved the issue and now it works for sure on Windows and Linux. Both Unity and a lots of other games and sims.

MHeironimus, 2022-03-30:
Version 2.1.0 of the library should address the issues seen in Unity (and other applications).

closed 2022-03-30 by MHeironimus