MHeironimus/ArduinoJoystickLibrary #105: Clone compatible?

ChicoCybergoth, 2019-01-10:
This clone is compatible?
https://www.ebay.com/itm/New-Pro-Micro-ATmega32U4-5V-16MHz-Replace-ATmega328-Pro-Mini-for-Arduino/162857904761
Very good price.

Thank you.

bobbydilley, 2019-01-30:
Without actually purchasing one, the developer of this software has no more insight into if this would work than you do. Seeing as it's a clone of a Pro Micro, and the software says it would work with a Pro Micro I would bet that it probably will work, but there is only one way to find out... ;)

ChicoCybergoth, 2019-01-30:
It is cheap and you made me more confident, I will try!
Thank you very much.

hindenbugbite, 2019-02-25:
I am using a Pro Micro clone off ebay and it works fine so far. I did reflash the arduino bootloader because I didn't see any com ports show up at first.

Enchiridion, 2020-02-26:
@ChicoCybergoth I'm using a cheap clone from Aliexpress and it works great.

ChicoCybergoth, 2020-03-18:
@Enchiridion Thank you for the information!
[image]
This one?

Enchiridion, 2020-03-18:
@ChicoCybergoth Yes, it looks exactly like that one.

helpme2k2, 2020-04-28:
> 
> 
> I am using a Pro Micro clone off ebay and it works fine so far. I did reflash the arduino bootloader because I didn't see any com ports show up at first.

@hindenbugbite when you said you flash the Arduino bootloader onto the Pro Micro clone board, which bootloader are you referring to? the Arduino Micro? I have a clone board, exactly the same as the one in picture and planning to use it to build a joystick. But I am having so much trouble getting the board to work the the IDE. Windows just doesn't want to recognize the board, double reset is a hit or miss. I wonder another bootloader will help.

hindenbugbite, 2020-04-28:
@helpme2k2 I did flash an Arduino Micro bootloader to it. I used another Arduino Uno as the ISP programmer as described here: https://www.arduino.cc/en/tutorial/arduinoISP Once the bootloader is loaded, it should show up in windows as a com port. Then Arduino IDE can load sketches.

helpme2k2, 2020-04-28:
@hindenbugbite Wow, thanks for the prompt reply. After the flash, in the IDE , do you choose the board type as Arduino Micro? or Pro Micro?

hindenbugbite, 2020-04-29:
I would use Pro Micro since it might be better defined for the target. They are essentially the same with the Pro Micro being a subset of the Micro.

helpme2k2, 2020-04-30:
@hindenbugbite I have an official Leonardo and it works perfectly on my windows 10 pc through a USB 3.0 port. Then I bough this Pro Micro from China and it is so unreliable, the serial port can only be detected 1 out of  20 times. I tried switching cables and ports, but it gives the same result. I even flashed the bootloader, thanks to your help) May I know if Pro Micro supposed to behave the same as Leonardo?

closed 2020-07-01 by MHeironimus