MHeironimus/ArduinoJoystickLibrary #109: Problem with a buttonbox

Bepy93, 2019-02-18:
Hello Mr. Heironimus! First of all, thank's very much for your wonderful library! I've created a buttonbox sketch to use with a flight simulator with 32 non-momentary switches. But i have a problem: i need to change the operation of some switches, making them become pushbuttons. I've tried a lot, but i can't make them work as i want. Can you help me please Mr. Heironimus?

so, here we have 32 non-momentary switches, that give an imput every time there are switched ON or OFF. Suppose to use a pushbutton on Button #1, (Joy_button=2) how and where i must modify the sketch?

#include <Keypad.h>
#include <Joystick.h>

#define ENABLE_PULLUPS
#define NUMBUTTONS 32
#define NUMROWS 7
#define NUMCOLS 5

byte buttons[NUMROWS][NUMCOLS] = {
  {0, 1, 2, 3, 4},
  {5, 6, 7, 8, 9},
  {10, 11, 12, 13, 14},
  {15, 16, 17, 18, 19},
  {20, 21, 22, 23, 24},
  {25, 26, 27, 28, 29},
  {30, 31}
};

byte rowPins[NUMROWS] = {1, 2, 3, 4, 5, 6, 7};
byte colPins[NUMCOLS] = {8, 9, 10, 11, 12};

Keypad buttbx = Keypad( makeKeymap(buttons), rowPins, colPins, NUMROWS, NUMCOLS);

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,
                   JOYSTICK_TYPE_JOYSTICK, 32, 0,
                   false, false, false, false, false, false,
                   false, false, false, false, false);

void setup() {
  Joystick.begin();
}

void loop() {

  if (buttbx.getKeys())
  {
    if ( buttbx.key[0].stateChanged )
    {
      switch (buttbx.key[0].kstate) {                               //Button 1 working perfectly!
        case PRESSED:
          Joystick.setButton(buttbx.key[0].kchar, 1);
          delay(100);
          Joystick.setButton(buttbx.key[0].kchar, 0);
          break;
        case RELEASED:
          Joystick.setButton(buttbx.key[0].kchar, 1);
          delay(100);
          Joystick.setButton(buttbx.key[0].kchar, 0);
          break;
      }
    }
    if ( buttbx.key[1].stateChanged )                            //Button 2 work like Button 1, i wanna to use this     
    {                                                                              //like a pushbutton
      switch (buttbx.key[1].kstate) {
        case PRESSED:
        case HOLD:
          Joystick.setButton(buttbx.key[1].kchar, 1);
          break;
        case RELEASED:
        case IDLE:
          Joystick.setButton(buttbx.key[1].kchar, 0);
          break;
      }
    }
    if ( buttbx.key[2].stateChanged )                         //Button 3 work like button 1 again
    {
      switch (buttbx.key[2].kstate) {
        case PRESSED:
        case HOLD:
          Joystick.setButton(buttbx.key[2].kchar, 1);
          break;
        case RELEASED:
        case IDLE:
          Joystick.setButton(buttbx.key[2].kchar, 0);
          break;
      }
    }
  }
}

arduino recognize ALL button like non momentary switches
Thank's a lot!!!

Giuseppe

Bepy93, 2019-02-26:
Resolved.

hvyhitter, 2020-01-13:
Created an account to say this isnt resolved. From the above example. The original code is this 

void CheckAllButtons(void) {
      if (buttbx.getKeys())
    {
       for (int i=0; i<LIST_MAX; i++)   
        {
           if ( buttbx.key[i].stateChanged )   
            {
            switch (buttbx.key[i].kstate) {  
                    case PRESSED:
                    case HOLD:
                              Joystick.setButton(buttbx.key[i].kchar, 1);
                              break;
                    case RELEASED:
                    case IDLE:
                              Joystick.setButton(buttbx.key[i].kchar, 0);
                              break;
            }
           }   
         }
     }

Now if you want to take a toggle switch that is either ON or off. but is not momentary. 

The code you replace that with is this. 

void CheckAllButtons(void) {
      if (buttbx.getKeys())
    {
       for (int i=0; i<LIST_MAX; i++)   
        {
           if ( buttbx.key[i].stateChanged )   
            {
            switch (buttbx.key[i].kstate) 
            {  
                    case PRESSED:
                              Joystick.setButton(buttbx.key[i].kchar, 1);
                              delay(100);
                              Joystick.setButton(buttbx.key[i].kchar, 0);
                              break;
                    case RELEASED:
                              Joystick.setButton(buttbx.key[i].kchar, 1);
                              delay(100);
                              Joystick.setButton(buttbx.key[i].kchar, 0);
                              break;
           }
           }
   

What this post was trying to say is that any switch a Arduino sees is seen as a pulse Even if the switch stays ON. so it will just do a pushbutton for a second. 

I have no idea how to make some buttons act like pushbuttons (stay on for the length you press them) and others act like toggles that come on only for a second when turned to the on position then when turned off press the button again. 

If I figure it out I will post it below. This is a cool project to make a button box. Ive been at this for 12 hours now and there is plenty of people searching for this info but no answers that are perfect.

DavDaddy, 2023-02-19:
This code should do the trick. I don't have a board to test this on that isn't in a project so it may not work.

Or now that I think about it, it may let you use a momentary switch as a toggle going on with the first press & off with the next. 

`  #include <Joystick.h>

Joystick_ Joystick;

const int BUTTON = 0;
bool switchState = false;

void setup() {
  Joystick.begin();
  Joystick.setButton(BUTTON, switchState);
}

void loop() {
  bool buttonState = digitalRead(BUTTON);
  if (buttonState == HIGH) {
    switchState = !switchState;
  }
  Joystick.setButton(BUTTON, switchState);
  Joystick.sendState();
  delay(50);
}
`