MHeironimus/ArduinoJoystickLibrary #113: 'ghost' input on the joystick

mikevromen, 2019-03-11:
## Description of Issue
Enter description of issue here. Include a clear and concise description of what you expected to happen and what actually happened.

When I enable X & Y axis in the contructor Windows detects the input (left/upperLeft) from the Joystick.

Since I only pullup one pin that is wired to a button (on/off) I don't expect windows to detect any joystick input.

## Technical Details

- Arduino Board (e.g. Arduino Leonardo): Due
- Host OS (e.g. Windows 10): Windows 10 / macOS / Xbox One OS (with adaptive controller)
- Arduino IDE Version (e.g. 1.8.3): 1.8.8

## Sketch File that Reproduces Issue

~~~
#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD,
  4, 0,                  // Button Count, Hat Switch Count
  true, true, false,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering

void setup() {
  // Initialize Button Pins
  pinMode(47, INPUT_PULLUP);
  pinMode(48, INPUT_PULLUP);
  pinMode(49, INPUT_PULLUP);
  pinMode(50, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin();

  Joystick.setYAxis(0);
  Joystick.setXAxis(0);
  Joystick.setZAxis(0);

  Joystick.setYAxisRange(0,0);
  Joystick.setXAxisRange(0,0);
  Joystick.setZAxisRange(0,0);
}

// Constant that maps the phyical pin to the joystick button.
const int pinToButtonMap = 47;

// Last state of the button
int lastButtonState[4] = {0,0,0,0};

void loop() {
  Joystick.setYAxis(0);
  Joystick.setXAxis(0);
  Joystick.setZAxis(0);

  Joystick.setYAxisRange(0,0);
  Joystick.setXAxisRange(0,0);
  Joystick.setZAxisRange(0,0);
  // Read pin values
  for (int index = 0; index < 4; index++)
  {
    int currentButtonState = !digitalRead(index + pinToButtonMap);
    if (currentButtonState != lastButtonState[index])
    {
      Joystick.setButton(index, currentButtonState);
      lastButtonState[index] = currentButtonState;
    }
  }

  delay(50);
}

~~~

## Wiring Details
Any pin wiring details that may be relevant.

4 buttons, each wired to a digitalPin on the Arduino.
1 shared ground wired to a groundPin on the Arduino.
(using 1 button with only 1 ground results in the same issue)

## Additional context
So far I tried to disable the X and Y axis in the constructor, sadly this results in the xbox adaptive controller not picking up any input.

KajTH, 2019-03-25:
I have the exact same problem.

mikevromen, 2019-03-25:
Maybe important to note, this is also happening on a Arduino Micro.

KajTH, 2019-03-25:
Joystick.setXAxisRange(-127, 127);
Joystick.setYAxisRange(-127, 127);

You need to set the minimum range to -127

phwoelfel, 2019-03-25:
Why do you enable it if you don't want to use it? Just disable them in the constructor so they won't be detected. If you want the buttons to be left/right and up/down for the joystick axis you have to set the range to -1, 1 and then set the axis accordingly when the buttons are pressed.

mikevromen, 2019-03-26:
> Why do you enable it if you don't want to use it? Just disable them in the constructor so they won't be detected.
> 
> If you want the buttons to be left/right and up/down for the joystick axis you have to set the range to -1, 1 and then set the axis accordingly when the buttons are pressed.

It's fixed thanks to KajTH's suggestion.

And if I disable them in the constructor the Arduino will not be 'valid' input for the xbox adeptive controller. That needs a joystick to be detected.

closed 2019-03-26 by mikevromen