MHeironimus/ArduinoJoystickLibrary #114: Library speed

eleksis, 2019-03-16:
## Description of Issue
Hi, made a quick speed test of this library (look under sketch for example). And it turns out it takes quite a long time to send/update state (~250ms). 

Are there any options to speed it up? Because for game controller you want to have as quick "response time" as possible. This is not suitable for any serious gamming ( should be at least under 50ms).

I didn't dive deep into code yet, but noticed sendState() updates all axes every time. Even if i am using just one. Or is the buttleneck somewhere else? Can you give some guidence please.

## Technical Details

- Arduino Pro Micro
- Linux/Win10
- Arduino IDE Version 1.8.8

## Sketch File that Reproduces Issue
```
unsigned long timer = 0;
void loop() {
  Serial.println(millis() - timer );
  timer = millis();  
  Joystick.setZAxis(100);
}
```

eleksis, 2019-03-16:
Update: setButton is fast enough. The problem is obviously with setAxis

jakubdybczak, 2019-10-18:
I came across same issue. This is OS depended (OS doesn't update controller state frequently when there is no app using this controller). On Linux I had use this command:
`cat /dev/input/js0`
in order to make loop duration fast

eleksis, 2020-01-17:
@jakubdybczak Thanks for info. Closing this issue.

closed 2020-01-17 by eleksis