MHeironimus/ArduinoJoystickLibrary #116: Analog shake trouble

sergiopoverony, 2019-05-23:
## Description of Issue
Enter description of issue here. Include a clear and concise description of what you expected to happen and what actually happened.
After download ino to Arduino Micro/Leonardo in gamepad test i see as analog poit noise (all shake) - analog joypad not connected, after connect analog joypad - the problem is not solved
[image]

## Technical Details
- Arduino Micro 
- Windows 10
- Arduino IDE Version 1.8.5 and 1.8.9

## Sketch File that Reproduces Issue

~~~
#include <Joystick.h>
#include <Arduino.h>
#include <DynamicHID.h>

#define PINS 16
#define ENABLE_ANALOG1 true
int X1 = A0;
int Y1 = A1;

// Create Joystick
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_GAMEPAD, PINS, 0, 
  true, true, false, false, false, false, false, false, false, false, false);

class CButton {
  public:
  int pin = 0;
  int lastState = 0;
  
  CButton(int p) {
    pin = p;
  }
};

CButton Buttons[PINS] ={A2,A3,0,1,2,3,4,5,6,7,8,9,10,16,14,15};

void setup() {
  for(int i=0 ; i<PINS ;i++) {
    pinMode(Buttons[i].pin, INPUT_PULLUP);
  }

  Joystick.begin();
  if (ENABLE_ANALOG1) {
    Joystick.setXAxisRange(-512, 512);
    Joystick.setYAxisRange(-512, 512);
  }
}

void JButtonStates() {
  if (ENABLE_ANALOG1) {
    Joystick.setXAxis(analogRead(X1) - 512);
    Joystick.setYAxis(analogRead(Y1) - 512);
  }
  
  for (int i = 0; i < PINS; i++) {
    int currentState = !digitalRead(Buttons[i].pin);
    
    if (currentState != Buttons[i].lastState) {
      Joystick.setButton(i, currentState);
      Buttons[i].lastState = currentState;
    }
  }  
}

void loop() {
  JButtonStates();
  delay(50);
}
~~~

## Wiring Details
analog point any time shaked and not work 
[image]

## Additional context
Add any other context about the problem here.

targodan, 2019-05-23:
The software looks good. I guess it's a wiring issue. Can you show/tell us how you wired up the Joystick?

sergiopoverony, 2019-05-23:
> The software looks good. I guess it's a wiring issue. Can you show/tell us how you wired up the Joystick?
Joystick not conencted. i only download sketch

fernandohsilva, 2019-05-23:
Maybe library not found!

targodan, 2019-05-23:
If the joystick is not connected this shaking is normal. Then the analog pins are floating.

sergiopoverony, 2019-05-23:
> Maybe library not found! 

w/out library he not compeled

sergiopoverony, 2019-05-23:
> If the joystick is not connected this shaking is normal. Then the analog pins are floating.

I connect this analog. Use correctly pins. Trouble not solved
https://datasheet.lcsc.com/szlcsc/ALPS-Electric-RKJXY1000006_C219776.pdf

targodan, 2019-05-23:
Can you please show us *how* you connected it? Maybe a schematic, or at least a photo?

sergiopoverony, 2019-05-23:
This hardy 

[image]
[image]

sergiopoverony, 2019-05-24:
Analog joypad diagram standarted as any analog. 
[image]

targodan, 2019-05-24:
Sorry, as I was trying to reply yesterday, my internet died.

Well, that looks all right. I don't know what would cause the shaking then. Maybe a bad connection on the header. Or a broken joystick module.

Can you try to verify the joystick module is OK? Maybe using a multimeter or an oscilloscope.

Also I don't see any capacitor near the atmega. Maybe try to solder in a 10nF across Vcc and GND of the Atmega to rule out stability problems.

sergiopoverony, 2019-05-24:
> Sorry, as I was trying to reply yesterday, my internet died.
>
> Well, that looks all right. I don't know what would cause the shaking then. Maybe a bad connection on the header. Or a broken joystick [...]

Joystick worker and contacts too. Tested with a multimeter. Tonight I will put a capacitor in for verification and I will write later.

sergiopoverony, 2019-05-25:
IDK what need change. My Analog resistance  2Kom. Any other  10Kom
maybe who know?>

byteborg, 2020-10-10:
Looks like your catching noise on the lines. You could interpolate the measurements, as I do in one of my projects:
```C++
// Measure analog pin and return tare-corrected value
int64_t measure(uint8_t pin, uint8_t interpol, int64_t tare) {
    int64_t val = 0;

    for (uint8_t i = 0; i < interpol; i++) {
        val += analogRead(pin);
    }
    return (val / interpol) - tare;
}
```